platformio/platformio-core #358: Error: nucleo_f401re board doesn't support eth library!

andriyadi, 2015-12-07:
Hi,

I'm trying to use [ESP8266Interface](https://developer.mbed.org/teams/ESP8266/code/ESP8266Interface/) library in my mbed project, using nucleo_f401 board. And I got this message when building:
```Error: nucleo_f401re board doesn't support eth library!```

Apparently, this script that causes that: 
`platformio/platformio/builder/scripts/frameworks/mbed.py`

In the library, there's a class called `TCPSocketConnection`. I try to rename the file (.h and .cpp) to `TCPSocketConn` and that error ceases to exist. But I don't think that should be the solution, as I need to always rename the class for the new projects.

Please advice. Thanks.

ivankravets, 2015-12-07:
Please [install develop version](http://docs.platformio.org/en/latest/installation.html#d-development-version) and wait for the further steps from @valeros.

closed 2015-12-07 by ivankravets

valeros, 2015-12-07:
Hi @andriyadi 
`ESP8266Interface` library currently in beta and has a little bit unusual structure.
I can offer your working project with some modifications:
https://dl.dropboxusercontent.com/u/10025728/ESP8266Interface.zip

andriyadi, 2015-12-07:
I can see that the `develop` version now offers warning, instead of error.
```Warning: nucleo_f401re board doesn't have native support for 'eth' library!```

But build is still error like this:
```~/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: cannot find -leth```

I'll try @valeros solution.

valeros, 2015-12-07:
@andriyadi 
`cannot find -leth`
Please, reinstall develop version.

andriyadi, 2015-12-07:
Hi @valeros 
I can confirm that the latest `develop` version works!
Thanks!

commit a483ecdf (2015-12-07): Warn about missing native support for the library // Resolve #358