platformio/platformio-core #359: Using the serial ports monitor from within CLion

orome, 2015-12-07:
How do I use the serial ports monitor from within CLion? 

If I add an external tool and run it I get:

```
/usr/local/bin/platformio serialports monitor
--- Miniterm on /dev/cu.usbmodem1421: 9600,8,N,1 ---
--- Quit: Ctrl+]  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Error: (25, 'Inappropriate ioctl for device')
```

ivankravets, 2015-12-07:
Please use special plugin for it https://plugins.jetbrains.com/plugin/8031?pr=clion

orome, 2015-12-08:
Is there no way to get `platformio serialports monitor` working from within the IDE (perhaps with some configuring or arguments). I like using "External Tools" for things.

itsjustvenky, 2015-12-09:
@orome  Use "Terminal" from Clion which is already available. Check the ICON in botton left of your ide.

orome, 2015-12-09:
@itsjustvenky — Yes, understood. But what I was hoping for was a way to use it alongside other "External Tools" in my workflow.

ivankravets, 2015-12-09:
@orome I don't see in my CLion IDE "External tools" However, I see that exists documentation for it https://www.jetbrains.com/clion/help/external-tools.html

Could you provide me screenshot what do you mean under it? 

P.S: I use trial CLion.

orome, 2015-12-09:
I'd like to be able to do something like this:

[image]

Also, while we're at it. It would be great if somehow closing the window it's running in would turn it off/exit. The serial monitor plugin doesn't do that and is therefore almost useless (too much clicking and too easy to forget and leave it connected).

ivankravets, 2015-12-09:
PlatformIO uses @PySerial for `platformio serialports` command. [Miniterm](https://pythonhosted.org/pyserial/examples.html#miniterm) is used for Serial Monitor. However, `miniterm` requires to be run within terminal applicaiton. I've just switched directly to @PySerial, see
```
/Volumes/SOFT/Projects/GitHub/platformio/origin/.tox/develop/bin/python -m serial.tools.miniterm /dev/cu.usbmodemFD1311
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/Volumes/SOFT/Projects/GitHub/platformio/origin/.tox/develop/lib/python2.7/site-packages/serial/tools/miniterm.py", line 866, in <module>
    main()
  File "/Volumes/SOFT/Projects/GitHub/platformio/origin/.tox/develop/lib/python2.7/site-packages/serial/tools/miniterm.py", line 838, in main
    filters=filters)
  File "/Volumes/SOFT/Projects/GitHub/platformio/origin/.tox/develop/lib/python2.7/site-packages/serial/tools/miniterm.py", line 315, in __init__
    self.console = Console()
  File "/Volumes/SOFT/Projects/GitHub/platformio/origin/.tox/develop/lib/python2.7/site-packages/serial/tools/miniterm.py", line 123, in __init__
    self.old = termios.tcgetattr(self.fd)
termios.error: (25, 'Inappropriate ioctl for device')
```

See
> Miniterm itself does not implement any terminal features such as VT102 compatibility. However it inherits these features from the terminal it is run. 

Try to wrap PlatformIO to shell script, like
```shell
# !/bin/bash

/full/path/to/platformio $*
```

hpsaturn, 2018-09-06:
Hi,
 
In the last update of PlatformIO over Atom, I have a similar issue when I go to uploading:

``` python
...
Hash of data verified.
Compressed 204368 bytes to 104260...
Wrote 204368 bytes (104260 compressed) at 0x00010000 in 1.9 seconds (effective 855.3 kbit/s)...
Hash of data verified.
 
Leaving...
Hard resetting via RTS pin...
========================= [SUCCESS] Took 7.51 seconds =========================
Error: (25, 'Inappropriate ioctl for device')
```
but over CLI thats not happen with `pio run --target upload`:

``` python
...
Hash of data verified.
Compressed 204368 bytes to 104260...
Wrote 204368 bytes (104260 compressed) at 0x00010000 in 1.9 seconds (effective 846.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=================================================================== [SUCCESS] Took 6.76 seconds ===================================================================
```

ivankravets, 2018-09-06:
@hpsaturn could you reproduce this issue with https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide? I see that a lot of developers abandon Atom's extensions :(

hpsaturn, 2018-09-06:
Hi @ivankravets, that is a rally bad news for me. 😳😫😭. I don't like Microsoft products, my world is only with Linux, I prefer go back to vim, but I will try to test it and replicate it on visualstudio for you. Thanks my friend for quicky answer.

ivankravets, 2018-09-06:
VSCode != Microsoft Visual Studio. It is a separate free and open source project under MIT license. It is also based on the same Electron engine which Atom uses.

hpsaturn, 2018-09-06:
ok, I understand, I got confuse for microsoft link, ok, I will try it.

jlsalmon, 2019-02-13:
My slightly hacky but reasonably effective workaround for this is to call `platformio serialports monitor` as part of the CMake upload command, i.e. in CMakeLists.txt:

```
add_custom_target(
        PLATFORMIO_UPLOAD ALL
        COMMAND ${PLATFORMIO_CMD} -f -c clion run --target upload && sleep 1 && ${PLATFORMIO_CMD} serialports monitor
        WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)
```

ivankravets, 2019-02-23:
We added new targets in https://github.com/platformio/platformio-core/commit/e0b9d080fa099c2ead66867a48d039d4ae08a148

You can run Serial Port Monitor as a build task.

Please re-test with the latest development version via `$ pio upgrade --dev`. 

P.S: Don't forget to install https://plugins.jetbrains.com/plugin/7177-file-watchers plugin. See updated docs > https://docs.platformio.org/en/latest/ide/clion.html

closed 2019-02-23 by ivankravets

EmmanuelMess, 2023-02-20:
> We added new targets in https://github.com/platformio/platformio-core/commit/e0b9d080fa099c2ead66867a48d039d4ae08a148
> 
> You can run Serial Port Monitor as a build task.
> 
> Please re-test with the latest development version via $ pio upgrade --dev.
> 
> P.S: Don't forget to install [plugins.jetbrains.com/plugin/7177-file-watchers](https://plugins.jetbrains.com/plugin/7177-file-watchers) plugin. See updated docs > [docs.platformio.org/en/latest/ide/clion.html](https://docs.platformio.org/en/latest/ide/clion.html)

Still doesn't work.

I am getting:

```
/home/emmanuel/.platformio/penv/bin/platformio -c clion device monitor -e uno
Error: Traceback (most recent call last):
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/platformio/cli.py", line 70, in invoke
    return super().invoke(ctx)
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/platformio/device/monitor/command.py", line 152, in device_monitor_cmd
    start_terminal(options)
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/platformio/device/monitor/terminal.py", line 54, in start_terminal
    term = new_terminal(options)
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/platformio/device/monitor/terminal.py", line 102, in new_terminal
    term = Terminal(
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/platformio/device/monitor/terminal.py", line 28, in __init__
    super().__init__(*args, **kwargs)
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/serial/tools/miniterm.py", line 395, in __init__
    self.console = Console()
  File "/home/emmanuel/.platformio/penv/lib/python3.8/site-packages/serial/tools/miniterm.py", line 198, in __init__
    self.old = termios.tcgetattr(self.fd)
termios.error: (25, 'Inappropriate ioctl for device')

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

divStar, 2023-06-14:
Same here, I found an issue on CLion for this and posted my findings there (I am trying to use the built-in PlatformIO Serial Monitor in CLion): https://youtrack.jetbrains.com/issue/CPP-19591/CLion-Tools-PlatformIO-Monitor-Error-25-Inappropriate-ioctl-for-device .

I am getting a very similar error message as @EmmanuelMess .

commit e0b9d080 (2019-02-23): Project Generator: add new targets for CLion IDE "BUILD_VERBOSE" and "MONITOR" (serial port monitor) // Resolve #359
commit d2c2171e (2019-02-23): Project Generator: add new targets for CLion IDE "BUILD_VERBOSE" and "MONITOR" (serial port monitor) // Resolve #359