platformio/platformio-core #360: ImportError: cannot import name LockFile

QuimeraCoding, 2015-12-08:
Hello.

I've trying to use platformio as it looks like a wonderfull tool. However, Altough it installs apparently with success, running platformio allways shows the same errors:

Traceback (most recent call last):
  File "/usr/local/bin/platformio", line 7, in <module>
    from platformio.__main__ import main
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 24, in <module>
    from platformio import __version__, exception, maintenance
  File "/usr/local/lib/python2.7/dist-packages/platformio/maintenance.py", line 24, in <module>
    from platformio import __version__, app, exception, telemetry
  File "/usr/local/lib/python2.7/dist-packages/platformio/app.py", line 21, in <module>
    from lockfile import LockFile
ImportError: cannot import name LockFile

I've tried installing from the script, with pip and with get-platformio.py, but maybe I'm missing something..
I'ts too bad I can't seem to use your tool, it seems just what I was looking for.
Thank you in advance for your time.

I've tried on Ubuntu 14.04 and Linux Arm, python 2.7 and pip 7.1.2

pip freeze:

adium-theme-ubuntu==0.3.4
apt-xapian-index==0.45
boto==2.38.0
bottle==0.12.9
chardet==2.0.1
click==5.1
command-not-found==0.3
configobj==5.0.6
crcmod==1.7
debtagshw==0.1
defer==1.0.6
dirspec==13.10
dnspython==1.11.1
duplicity==0.6.23
enum34==1.1.1
glob2==0.4.1
google-api-python-client==1.4.2
httplib2==0.9.2
idna==2.0
ipaddress==1.0.15
Jinja2==2.8
lockfile==0.12.2
lxml==3.3.3
MarkupSafe==0.23
mysql-connector-python==1.1.6
mysql-utilities==1.3.5
oauth2client==1.5.2
oauthlib==0.6.1
oneconf==0.3.7.14.4.1
ordereddict==1.1
PAM==0.4.2
paramiko==1.10.1
pexpect==3.1
Pillow==2.3.0
piston-mini-client==0.7.5
platformio==2.4.1
pyasn1==0.1.9
pyasn1-modules==0.0.8
PyBluez==0.18
pycparser==2.14
pycrypto==2.6.1
pycups==1.9.66
pycurl==7.19.3
pygobject==3.12.0
pyOpenSSL==0.13
pyserial==2.7
pysmbc==1.0.14.1
pysqlite==2.6.3
python-apt===0.9.3.5ubuntu1
python-dateutil==2.4.2
python-debian===0.1.21-nmu2ubuntu2
python-magic==0.4.10
pyxdg==0.25
reportlab==3.0
requests==2.8.1
rsa==3.2.3
s3cmd==1.6.0
sessioninstaller==0.0.0
simplejson==3.8.1
six==1.10.0
software-center-aptd-plugins==0.0.0
system-service==0.1.6
Twisted-Core==13.2.0
Twisted-Web==13.2.0
unity-lens-photos==1.0
uritemplate==0.6
urlgrabber==3.9.1
urllib3==1.7.1
vboxapi==1.0
virtualenv==13.1.2
wheel==0.26.0
xdiagnose===3.6.3build2
yum-metadata-parser==1.1.4
zope.interface==4.0.5

ivankravets, 2015-12-08:
I don't see `setuptools` package in your list. Try
```
pip uninstall lockfile
pip install -U setuptools pip
pip install lockfile
```

QuimeraCoding, 2015-12-08:
Hi Ivan, thank you so much for your quick response.

I've tried the steps you mention above but with no success, still shows the exact same error.
Maybe it's my systems problem, I have to do my pip install's with sudo -H, but don't quite think it's the problem. 

Although pip freeze doesn't show setuptools package (wich was installed and I re-installed), pip list does, as shown next:

adium-theme-ubuntu (0.3.4)
apt-xapian-index (0.45)
boto (2.38.0)
bottle (0.12.9)
chardet (2.0.1)
click (5.1)
command-not-found (0.3)
configobj (5.0.6)
crcmod (1.7)
debtagshw (0.1)
defer (1.0.6)
dirspec (13.10)
dnspython (1.11.1)
duplicity (0.6.23)
enum34 (1.1.1)
glob2 (0.4.1)
google-api-python-client (1.4.2)
httplib2 (0.9.2)
idna (2.0)
ipaddress (1.0.15)
Jinja2 (2.8)
lockfile (0.12.2)
lxml (3.3.3)
MarkupSafe (0.23)
mysql-connector-python (1.1.6)
mysql-utilities (1.3.5)
oauth2client (1.5.2)
oauthlib (0.6.1)
oneconf (0.3.7.14.4.1)
ordereddict (1.1)
PAM (0.4.2)
paramiko (1.10.1)
pexpect (3.1)
Pillow (2.3.0)
pip (7.1.2)
piston-mini-client (0.7.5)
platformio (2.4.1)
pyasn1 (0.1.9)
pyasn1-modules (0.0.8)
PyBluez (0.18)
pycparser (2.14)
pycrypto (2.6.1)
pycups (1.9.66)
pycurl (7.19.3)
pygobject (3.12.0)
pyOpenSSL (0.13)
pyserial (2.7)
pysmbc (1.0.14.1)
pysqlite (2.6.3)
python-apt (0.9.3.5ubuntu1)
python-dateutil (2.4.2)
python-debian (0.1.21-nmu2ubuntu2)
python-magic (0.4.10)
pyxdg (0.25)
reportlab (3.0)
requests (2.8.1)
rsa (3.2.3)
s3cmd (1.6.0)
sessioninstaller (0.0.0)
setuptools (18.7.1)
simplejson (3.8.1)
six (1.10.0)
software-center-aptd-plugins (0.0.0)
system-service (0.1.6)
Twisted-Core (13.2.0)
Twisted-Web (13.2.0)
unity-lens-photos (1.0)
uritemplate (0.6)
urlgrabber (3.9.1)
urllib3 (1.7.1)
vboxapi (1.0)
virtualenv (13.1.2)
wheel (0.26.0)
xdiagnose (3.6.3build2)
yum-metadata-parser (1.1.4)
zope.interface (4.0.5)

ivankravets, 2015-12-08:
You have mixed installation. Some part of packages is globally installed where another part is installed to home directory. You should decide where you want to install PlatformIO.

What is more, do you have installed this package in your system? https://launchpad.net/ubuntu/+source/python-lockfile

Finally, try to install PlatformIO into virtual environment:

```
cd /tmp
virtualenv pio
source pio/bin/activate
pip install platformio
```

QuimeraCoding, 2015-12-08:
Ok, It's now running in Ubuntu 14.04 and Linux ARM into virtual environment, will try to compile and upload later.

The package python-lockfile was on the latest version but the installation into virtual environment went ok, with no issues.

Guess this is all due to my inexperience with python, sorry for that.

So now, when ever I want to run platformIO I must setup virtualenv, is that correct?

I'm working on a little project to manage IoT setups so that anyone can connect their board to the usb port and in a few minutes have it connected to the system and get an instant graphical output on the web, they connect it we code it. Allready prepared a MQTT broker on Node.js and now I'm setting up the cloud compile mechanism for which platformIO seems the perfect. 

I've tried Inotool but I guess it has had better days.

I'll read a bit more about python and virtualenv so to not create more "false" issues.

Again, thank you for your time and your great work!

ivankravets, 2015-12-08:
> So now, when ever I want to run platformIO I must setup virtualenv, is that correct?

No, this is temporary solution for your issue. As I said above, you need to decide where you will install 3-rd party packages.

Try to uninstall `python-lockfile` package using `apt-get`. Then run this command `pip install -U platformio`.

ivankravets, 2015-12-08:
> I'm working on a little project to manage IoT setups so that anyone can connect their board to the usb port and in a few minutes have it connected to the system and get an instant graphical output on the web, they connect it we code it. Allready prepared a MQTT broker on Node.js and now I'm setting up the cloud compile mechanism for which platformIO seems the perfect tool.

Sounds very interesting. Where I can look on it? @PlatformIO was initially developed for my 2-nd project @smartanthill.

Don't forget to use `-f/--force` option before each `platformio` command when run it in SUBPROCESS. More details here [PlatformIO blocks command execution using user prompt](http://docs.platformio.org/en/latest/faq.html#platformio-blocks-command-execution-using-user-prompt)

QuimeraCoding, 2015-12-09:
Worked like a charm!

Both on Ubuntu and on ARM. Even though in Ubuntu got: Error: Invalid installation of Python `requests` package`. See < https://github.com/platformio/platformio/issues/252 >, the link sent me straight to the final step.

Tested with wire-blink example from Raspberry to Arduino Uno, and also with a sketch of my own, imported the libraries with no problems and it seems to be all up and running ok.

I'm guessing @PlatformIO will be my tool of choice to manage the libraries and compilations, it seems intuitive, dynamic, "clean" and objective, nice work indeed. @smartanthill seems very interesting too, will check out again with more attention in a few days!

As to my small project, although I have a very embrionic and raw structure online, I have no reasonable interface yet, will set up some proto-frontend some time next month. For now, most of the debugging is made locally with some Arduinos connecting to the cloud with MQTT over websockets and a Raspberry pi with a Node.js server with an embedded MQTT broker bridging to the clone server/broker in the cloud.

It's called iotopia.io, and soon as I have something to "show" I'll sent you a note. It's going slow but steady, because it's something I do in my free time, in my "day job" I help to develop an ERP for a software company. I most certainly would like to spend more time developing for IoT but for now it has to be a few hours here and there.

Again, thank you so much for the help and your readiness, I will be contacting you one of this days.

ivankravets, 2015-12-09:
Thanks! Will wait news from you. I close this issue.

closed 2015-12-09 by ivankravets

glynhudson, 2016-06-05:
I have the same issue `ImportError: cannot import name LockFile`

When I try and run `sudo pip uninstall lockfile` I get errror: `Not uninstalling lockfile at /usr/lib/python2.7/dist-packages, owned by OS`

```
Nautilus-scripts-manager==1.7
PAM==0.4.2
Pillow==2.3.0
Pmw==1.3.2
Twisted-Core==13.2.0
Twisted-Web==13.2.0
adium-theme-ubuntu==0.3.4
apt-xapian-index==0.45
argparse==1.2.1
backports-abc==0.4
backports.ssl-match-hostname==3.5.0.1
bottle==0.12.9
certifi==2016.2.28
chardet==2.0.1
click==5.1
colorama==0.2.5
command-not-found==0.3
debtagshw==0.1
defer==1.0.6
derpconf==0.7.3
dirspec==13.10
dnspython==1.11.1
duplicity==0.6.23
esptool==0.1.0
futures==3.0.5
gyp==0.1
html5lib==0.999
httplib2==0.8
libthumbor==1.3.1
lockfile==0.8
lxml==3.3.3
numpy==1.8.2
oauthlib==0.6.1
oneconf==0.3.7.14.04.1
pexif==0.15
pexpect==3.1
piston-mini-client==0.7.5
platformio==2.9.4
pyOpenSSL==0.13
pycrypto==2.6.1
pycups==1.9.66
pycurl==7.19.3
pygobject==3.12.0
pyserial==2.6
pysmbc==1.0.14.1
python-apt==0.9.3.5ubuntu2
python-debian==0.1.21-nmu2ubuntu2
python-magic==0.4.11
pytz==2016.4
pyxdg==0.25
reportlab==3.0
requests==2.2.1
sessioninstaller==0.0.0
singledispatch==3.4.0.3
six==1.5.2
software-center-aptd-plugins==0.0.0
statsd==3.2.1
system-service==0.1.6
tornado==4.3
unity-lens-photos==1.0
uptime==3.0.1
urllib3==1.7.1
wheel==0.24.0
wsgiref==0.1.2
wxPython==3.0.1.1
wxPython-common==3.0.1.1
xdiagnose==3.6.3build2
xmltodict==0.10.1
zope.interface==4.0.5
```

ivankravets, 2016-06-05:
You need to use apt 
```
apt uninstall python-lockfile
```

Then `pip install platformio`

glynhudson, 2016-06-06:
Thanks @ivankravets 

However I needed:

`sudo apt-get remove python-lockfile`

I also needed to remove python requests, follow this guide did the trick: https://github.com/platformio/platformio/issues/252/.

Thanks a lot. I'm now up and running 👍, and probably know enough to be dangerous :-)

delkon, 2017-08-22:
Solution is:

```
sudo apt-get remove python-serial
sudo pip install pyserial
```