platformio/platformio-core #361: Proper way to import mbed library

andriyadi, 2015-12-08:
Hi again guys,

I wonder if there's a proper way to import or include libraries in mbed portal into platformio project, say using IDE CLion.

What I'm doing currently is "exporting to desktop IDE" program and libraries from mbed portal, and copy and paste to lib folder. It would be awesome if the `platformio lib` command is able to accommodate that, or other options.

Platformio is getting integrated deeply into my workflow. Thanks for all the efforts!

ivankravets, 2015-12-08:
PlatformIO Library Registry contains the popular mbed libraries. See http://platformio.org/#!/lib/search?query=framework%253Ambed&page=1.

Is your library registered here?

ivankravets, 2015-12-10:
Please reopen if you still need help.

andriyadi, 2015-12-11:
Sorry I missed this. 
I'd like to integrate [HTTPClient-SSL libraries](https://developer.mbed.org/teams/MultiTech/code/HTTPClient-SSL/), as it seems working flawlessly with ESP8266Interface. Also [VS1053Lib](https://developer.mbed.org/users/ollie8/code/VS1053lib/) library or alike.

Please advice.

ivankravets, 2015-12-11:
Please place these libraries to `lib` folder into your project. If them work, then we will add it to PlatformIO Library registry

andriyadi, 2015-12-15:
HTTPClient-SSL libraries cannot be built by default, unless I change the file structure to be flat.
VS1053Lib is fine. I also managed to use this one, [VS1053b](https://developer.mbed.org/users/christi_s/code/VS1053b/)

andriyadi, 2015-12-20:
Hi @ivankravets 
I have not seen the VS1053b library when I search using `platformio lib search`

No pressure though :)

ivankravets, 2015-12-21:
@gandy92 could you update MBED library configs and include requested libs?

Thanks in advance!

dh1tw, 2016-01-02:
+1 for a import of mbed Libraries. I'm currently working with the MQTT client and find it rather cumbersome to download, copy them in the lib directory and adopt the folder structure so that the dependency finder can actually find them.

ivankravets, 2016-01-02:
@dh1tw you can pull request to https://github.com/platformio/platformio-libmirror/tree/master/configs/mbed with required library. I'll register it.

ivankravets, 2016-07-15:
@dh1tw  
>  I'm currently working with the MQTT client and find it rather cumbersome to download, copy them in the lib directory and adopt the folder structure so that the dependency finder can actually find them.

Could you specify the link to this library?

commit 758396c9 (2016-08-02): PlatformIO Library Manager 3.0 // Resolve #588 , Resolve #414, Resolve #498, Resolve #475, Resolve #410, Resolve #461, Resolve #361, Resolve #507, Resolve #554
closed 2016-08-08 by ivankravets