platformio/platformio-core #362: Avoiding IDE import errors

orome, 2015-12-08:
In my IDE (CLion 1.2.1) I have
```
#include <EEPROM.h>
```
which is reported by the IDE inspector as an error, along with usages of all the values it defines. Building with PlatformIO works fine (so I have this somewhere) but the IDE's file inspection can't locate it.

How do I ensure that CLion knows where to look for these headers?

ivankravets, 2015-12-08:
1. `platformio upgrade` to the latest 2.5.0 release
2. Update your project with used libraries using `platformio init --ide clion --board ???`

See warning in documentation http://docs.platformio.org/en/latest/ide/clion.html#project-generator
> The libraries which are added, installed or used in the project after generating process wont be reflected in IDE. To fix it you need to reinitialize project using `platformio init` (repeat it).

orome, 2015-12-08:
@ivankravets — So I need to specify the IDE and the board all over again. That seems wrong. Also, when I do that, some of the IDE settings get wiped out (e.g., window positions). Is there no way to just update eat project other than reinitializing it from scratch?

ivankravets, 2015-12-09:
> Is there no way to just update eat project other than reinitializing it from scratch?

You can clone your project, apply `platformio ini --ide clion`, then copy `CMakeLists.txt` to original project. 

I've just opened new issue/feature for it #364

closed 2015-12-09 by ivankravets

orome, 2015-12-09:
Cool, thanks I'll watch #364.