platformio/platformio-core #363: What does the 'arduino' avrdude programmer do?

orome, 2015-12-09:
PlatformIO (at least using CLion) executes 

    avrdude ... -c arduino ...

to load program the my Arduino Uno, but it's unclear to me what it's doing. All the documentation I've found refers to the value of the `-c` setting as the "programmer"; and all the documentation I can find about that term describe something that's a hardware component connected to the "ISCP header". But I'm connected over a serial port with no special hardware.

Is PlatformIO using the bootloader to program my Arduino? In that case what role does the `-c arduino` play (is it specifying the bootloader, or something that depends on how I'm connected)?

---
    avrdude -v -p atmega328p -C "~/.platformio/packages/tool-avrdude/avrdude.conf" -c arduino -b 115200 -P /dev/cu.usbmodem1421 -D -U flash:w:.pioenvs/uno/firmware.hex:i

itsjustvenky, 2015-12-09:
This is what I see from arduino IDE

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega328p -carduino -PCOM3 -b115200 -D -Uflash:w:C:\Users\VENKAT~1.VEN\AppData\Local\Temp\build727352015918313731.tmp/sketch_dec09a.cpp.hex:i 

Here is the documentation for avrdude : http://www.nongnu.org/avrdude/user-manual/avrdude_4.html#Option-Descriptions

ivankravets, 2015-12-09:
@itsjustvenky thanks for the answer.

@orome you can control programmer-id/protocol using [upload_protocol](http://docs.platformio.org/en/latest/projectconf.html#upload-protocol) option.

closed 2015-12-09 by ivankravets

orome, 2015-12-09:
@itsjustvenky — So? That's the documentation I quoted in the question. For `arduino` its explanation is 	"Arduino". I was looking for a bit more than that.

@itsjustvenky — The question is simply: What does choosing `-c arduino` do? Specifically, ***is it using the bootloader***, when I'm connected to the Uno using the serial port?

itsjustvenky, 2015-12-09:
@orome Best place to post this question is "Arduino Forums".

anyway 
"arduino" is a configuration name.

Go through this file

https://raw.githubusercontent.com/adafruit/Trinket_Arduino_Linux/1.0.5/arduino-1.0.5/hardware/tools/avrdude.conf

orome, 2015-12-09:
@itsjustvenky — I'm a beginner. That file doesn't help much. What  "Arduino Forums" do you recommend?

itsjustvenky, 2015-12-09:
@orome https://forum.arduino.cc/

if you find a clear answer, please do post the answer here. thanks