platformio/platformio-core #365: esptool upload to IP address (OTA)

itsjustvenky, 2015-12-09:
using Arduino IDE I can upload a ESP sketch over OTA to a remote ESP using command 

python.exe C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/espota.py -i 192.168.0.105 -p 8266 -f C:\Users\VENKAT~1.VEN\AppData\Local\Temp\build1686115882947863985.tmp/BasicOTA.cpp.bin 

Is there a way to do this in platformio

thanks
Venkatesh

valeros, 2015-12-09:
Hi @itsjustvenky 
`espota.py` is located here: `c:\Users\User\.platformio\packages\framework-arduinoespressif\tools\espota.py`
Firmware binary file is located here: `path_to_project\.pioenv\env\firmware.bin`

ivankravets, 2015-12-09:
See here how to define own uploader http://docs.platformio.org/en/latest/projectconf.html#extra-script and related issue https://github.com/platformio/platformio/issues/351#issuecomment-161789165

itsjustvenky, 2015-12-10:
@valeros @ivankravets thank you.

itsjustvenky, 2015-12-10:
Sorry one more question : Can I set the value of properties like this "$UPLOADER" inside platformio.ini ? it seems to be not working. Also can I create a new properties like UPLOADER_IP_OTA  and use it in script ? please let me know. thanks.

ivankravets, 2015-12-10:
> Can I set the value of properties like this "$UPLOADER" inside platformio.ini 

No, you can't. The only way is extra script.

>  can I create a new properties like UPLOADER_IP_OTA and use it in script

You can create own macros and use it. For example,
```ini
# platformio.ini for the env
src_build_flags = -DUPLOADER_IP_OTA=HELLO

# extra_script
print env['CPPDEFINES'].get("UPLOADER_IP_OTA")
```

To print current environment of @PlatformIO builder use `platformio run --target envdump`

itsjustvenky, 2015-12-10:
@ivankravets  Thanks for the solution.

ivankravets, 2015-12-10:
Does it work for you?

itsjustvenky, 2015-12-10:
Nope.

I added lines exactly what you have given in above comments.

I used this command in script  : print env['CPPDEFINES'] 
and the output is

['F_CPU=$BOARD_F_CPU', '__ets__', 'ICACHE_FLASH', 'ARDUINO_ESP8266_ESP01', 'ARDUINO_ARCH_ESP8266', 'ESP8266', 'ARDUINO=200', ['UPLOADER_IP_OTA', 'HELLO'], 'PLATFORMIO=020500']

When I use this command : print env['CPPDEFINES'].get("UPLOADER_IP_OTA")
I am seeing this error : AttributeError: 'list' object has no attribute 'get':
----------------------
From above I can see "CPPDEFINES" is a list and with exception to one value "UPLOADER_IP_OTA", which is dictionary value.

please suggest. thanks.

ivankravets, 2015-12-10:
@itsjustvenky I got you what do you need. See updated https://github.com/platformio/platformio/issues/365#issuecomment-163695011

I didn't test it. Please report me if it doesn't work.

ivankravets, 2015-12-10:
I've just updated again my comment. Looks like you don't need to create own board. You can override LD script using [build_flags](http://docs.platformio.org/en/latest/projectconf.html#build-flags). 

If this works for you, I'll add it to PlatformIO and you can upload firmware using `--target uploadota --upload-port 127.0.0.1`

itsjustvenky, 2015-12-10:
I used the lines in script

'''
 UPLOADER=join("$PIOPACKAGES_DIR", "framework-arduinoespressif", "tools", "espota.py"),
 UPLOADCMD='$UPLOADER -i ${ENV.get("UPLOADER_IP_OTA")} -p 8266 -f $SOURCE'

I am getting error

'''
/home/venkat/.platformio/packages/framework-arduinoespressif/tools/espota.py -eo /home/venkat/.platformio/packages/framework-arduinoespressif/bootloaders/eboot/eboot.elf -bo .pioenvs/esp01-1m/firmware.bin -bm dio -bf 80 -bz 1M -bs .text -bp 4096 -ec -eo .pioenvs/esp01-1m/firmware.elf -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
Usage: espota.py [options]

espota.py: error: no such option: -e
scons: *** [.pioenvs/esp01-1m/firmware.bin] Error 2

-----------------------
So I changed the 

'''
UPLOADER_OTA=join("$PIOPACKAGES_DIR", "framework-arduinoespressif", "tools", "espota.py"),
    UPLOADCMD='$UPLOADER_OTA -i ${ENV.get("UPLOADER_IP_OTA")} -p 8266 -f $SOURCE'

UPLOADER ==> UPLOADER_OTA

and "espota.py" command (ota) works.

ivankravets, 2015-12-10:
See TOTALY updated comment. Please use that solution https://github.com/platformio/platformio/issues/365#issuecomment-163695011

itsjustvenky, 2015-12-10:
@ivankravets  Thank you so much. OTA works now with the configuration from platformio.ini. Appreciated!!!

ivankravets, 2015-12-10:
Thanks for the report. Ok, let's explain what have you changed. Do you use standard PlatformIO packages?

itsjustvenky, 2015-12-10:
I am still on SDK v1.4.. apart from this everything I reverted back.

ivankravets, 2015-12-10:
Where did you replace this SDK v1.4?

itsjustvenky, 2015-12-11:
I have complete overwritten these two folders with 1.4 SDK (esp-open-sdk).

```
~/.platformio/packages/sdk-esp8266
~/.platformio/packages/toolchain-xtensa
```

Here is the board I have created.

```
{
   "esp01-1m": {
        "build": {
            "core": "esp8266",
            "extra_flags": "-DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266",
            "f_cpu": "80000000L",
            "ldscript": "esp8266.flash.1m128.ld",
            "mcu": "esp8266",
            "variant": "generic"
        },
        "frameworks": ["arduino"],
        "name": "Espressif ESP8266 ESP-01-1MB board",
        "platform": "espressif",
        "upload": {
            "maximum_ram_size": 32768,
            "maximum_size": 1048576,
            "protocol": "arduino",
            "require_upload_port" : true,
            "speed": 115200
        },
        "url": "https://nurdspace.nl/ESP8266",
        "vendor": "Espressif"
    }
}
```

itsjustvenky, 2015-12-11:
Extra script for Arduino-OTA

```
from os.path import join
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

env.Replace(
    UPLOADEROTA=join("$PIOPACKAGES_DIR",
                     "framework-arduinoespressif", "tools", "espota.py"),
    UPLOADCMD='$UPLOADEROTA -i $UPLOAD_PORT -p 8266 -f $SOURCE --debug --progress'
)
```

I have added two other option's to command line, DEBUG and PROGRESS.

ivankravets, 2015-12-11:
> I have complete overwritten these two folders with 1.4 SDK (esp-open-sdk).
~/.platformio/packages/sdk-esp8266
~/.platformio/packages/toolchain-xtensa

Why did you replace these packages? `~/.platformio/packages/sdk-esp8266` isn't used in your example because Espressif Arduino contains own version of SDK See `~/.platformio/packages/framework-arduinoespressif/tools/sdk`.

Ok, please make these changes:
1. Backup `~/.platformio/boards` folder which you created for `esp01-1m`
2. Remove ``~/.platformio` directory
3. Run your example, PlatformIO should install stable frameworks, toolchains. Then place `boards` folder back.

Thanks!

itsjustvenky, 2015-12-11:
@ivankravets  Sorry for the confusion. Please ignore the SDK part. I reverted everything back to how it was installed by platformio (followed your above steps). thanks.

ivankravets, 2015-12-11:
> I reverted everything back to how it was installed by platformio (followed your above steps). thanks.

Does it work?

itsjustvenky, 2015-12-11:
Yes Arduino OTA works fine. 

One more thing : in file "~/.platformio/packages/sdk-esp8266/version.txt" I see
```
esp_iot_sdk_v1.2.0_15_07_03
```

But the actual SDK version @platformio provides is 1.3.. right ?

itsjustvenky, 2015-12-11:
If the "build_flags" is set in "platformio.ini", then it overrides the "ldscript" in custom boards json file.

ivankravets, 2015-12-11:
> But the actual SDK version @platformio provides is 1.3.. right ?

The actual is `esp_iot_sdk_v1.2.0_15_07_03`

> If the "build_flags" is set in "platformio.ini", then it overrides the "ldscript" in custom boards json file.

Yes

P.S: Wait from me new PlatformIO 2.5.1.dev0 where all these features are implemented.

itsjustvenky, 2015-12-11:
so you mean to say "esp_iot_sdk_v1.2.0_15_07_03" = 1.3 ?

itsjustvenky, 2015-12-11:
In my arduino (ide) sdk version it is "1.3.0_15_08_10_p1" (arduino esp8266 sdk 2.0.0)

ivankravets, 2015-12-11:
@itsjustvenky please don't mix `~/.platformio/packages/sdk-esp8266` and `~/.platformio/packages/framework-arduinoespressif/tools/sdk`. If you specify `framework = arduino` in `platformio.ini` then PlatformIO uses SDK from Espressif Arduino Framework, otherwise for native project will be used `~/.platformio/packages/sdk-esp8266`. See example https://github.com/platformio/platformio/tree/develop/examples/espressif/esp8266-native

closed 2015-12-11 by ivankravets

ivankravets, 2015-12-11:
I've just pushed new development version of PlatformIO 2.5.0.dev0. 

1. Please remove own `boards` folder from PIO home
2. Install development version http://docs.platformio.org/en/latest/installation.html#d-development-version
3. Remove from your project extra scripts and left only
```ini
[env:esp01_1m]
platform = espressif
framework = arduino
board = esp01_1m
upload_port = YOUR_DEVICE_IP_HERE
```

Waiting for the feedback. if you need additional flags to uploader see http://docs.platformio.org/en/latest/projectconf.html#upload-flags

itsjustvenky, 2015-12-11:
Installed dev version
```
venkat@venkat-VirtualBox:~$ platformio --version
PlatformIO, version 2.5.1.dev0
```
Test the Arduino-OTA and worked correctly.

Question : How do I revert back to 2.5.0 (release) ? if I want to

ivankravets, 2015-12-11:
> Test the Arduino-OTA and worked correctly.

Thanks!

> Question : How do I revert back to 2.5.0 (release) ? if I want to

```
pip uninstall platformio
pip install platformio
```

itsjustvenky, 2015-12-11:
```
venkat@venkat-VirtualBox:~/Projects/platformio/esp8266/esp01-1mdev$ platformio run  --target upload
[Fri Dec 11 20:47:10 2015] Processing esp01_1m (platform: espressif, upload_port: 192.168.0.105, board: esp01_1m, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/esp01_1m/firmware.bin"])
/home/venkat/.platformio/packages/framework-arduinoespressif/tools/espota.py --debug --progress -i 192.168.0.105 -f .pioenvs/esp01_1m/firmware.bin
20:47:11 [DEBUG]: Options: {'esp_ip': '192.168.0.105', 'image': '.pioenvs/esp01_1m/firmware.bin', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
20:47:11 [INFO]: Starting on 0.0.0.0:37848
20:47:11 [INFO]: Upload size: 226208
20:47:11 [INFO]: Sending invitation to: 192.168.0.105
20:47:11 [INFO]: Waiting for device...
Uploading: [============================================================] 100% Done...

20:47:16 [INFO]: Waiting for result...
20:47:16 [INFO]: Result: OK
============================================================================= [SUCCESS] Took 6.55 seconds =============================================================================
venkat@venkat-VirtualBox:~/Projects/platformio/esp8266/esp01-1mdev$ 
```

itsjustvenky, 2015-12-11:
@ivankravets thanks for all the help.

itsjustvenky, 2015-12-11:
Are you planning to release 2.5.1 anytime soon ?

ivankravets, 2015-12-11:
> Are you planning to release 2.5.1 anytime soon ?

See https://github.com/platformio/platformio/milestones . Release is planned on December 15, 2015

commit 2baf14f9 (2015-12-11): Implemented Over The Air (OTA) upgrades for Espressif development platform // Resolve #365