platformio/platformio-core #366: sdk-esp8266 update

celsworth, 2015-12-10:
Hello,

I'm starting to try some native SDK development for my esp8266 but the `sdk-esp8266` package is quite old: only `v1.2.0_15_07_03` :(

Could this be updated to 1.5 please?

sticilface, 2015-12-23:
I quite often use the latest git with the arduino IDE and the esp8266. Is this possible with platformio. The pace things move at is just so quick the releases can be quite far behind. I fact I can't find out how to know what release I am using in platformio?

by2coffee, 2015-12-23:
The toochain packages are stored in `~/.platformio/packages/`. To know your release version you (probably) can `cat ~/.platformio/packages/framework-arduinoespressif/version.txt`

sticilface, 2015-12-23:
Ah found it thanks.  Now the next question... is there an easy way ?script to convert the esp8266 github to a platformio compatible dir structure...

by2coffee, 2015-12-23:
If you mean the structure that uses folders like "user", "driver", and "include", then it's pretty straightforward since platformio's directory structure is pretty flexible (doesn't force much in terms of structure). In fact, if you just want to compile it, just cp and *c and *h files into the root `src` folder :stuck_out_tongue_closed_eyes: 

If you want to maintain the existing meaning you could try the following:

Please note that I **haven't tested this extensively** (actually, at all, sans two projects on my hard disk)

```sh
cd to/your/existing/project/folder/
PLATFORMIO_PRJ=/path/to/your/platformio/project
cp user/*.c include/*.h $PLATFORMIO_PRJ/src
mkdir -p $PLATFORMIO_PRJ/lib/driver/src
cp driver/*.c include/driver/*.h $PLATFORMIO_PRJ/lib/driver/src/
```

Of course, if you have more `MODULES` (other than `user` and `driver`) in your Makefile you'll have to repeat the last two lines with `driver` replaced by the module name

sticilface, 2015-12-23:
not sure i follow your instructions there.  

the version of SDK, plus the github code used for platformio is as mentioned the stable channel.  When developing I'm actually often using the latest github commit https://github.com/esp8266/Arduino with it linked into my arduino folder.

I've tried to link the various folders... cores, libraries, SDK etc... got almost there but it all fell over with the linking...  I provided the linker scripts from https://github.com/esp8266/Arduino, but it then fails at the end
```
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/firmware.elf -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -Wl,-T"/Users/amelvin/.platformio/packages/ldscripts/esp8266.flash.4m.ld" .pioenvs/nodemcu/src/Melvana.o .pioenvs/nodemcu/src/Melvanimate.o .pioenvs/nodemcu/src/Melvtrix.o .pioenvs/nodemcu/src/Palette.o .pioenvs/nodemcu/src/_E131.o .pioenvs/nodemcu/src/effectobject.o .pioenvs/nodemcu/src/tmp_ino_to.o -L/Users/amelvin/.platformio/packages/framework-arduinoespressif/tools/sdk/lib -L/Users/amelvin/.platformio/packages/ldscripts -L.pioenvs/nodemcu -Wl,--start-group -lsmartconfig -lpp -lmain -lwpa -llwip -lnet80211 -lwps -lcrypto -lphy -lhal -laxtls -lgcc -lm .pioenvs/nodemcu/libFrameworkArduinoVariant.a .pioenvs/nodemcu/libFrameworkArduino.a .pioenvs/nodemcu/libNeoPixelBus.a .pioenvs/nodemcu/libAdafruit_GFX_ID13.a .pioenvs/nodemcu/libJson_ID64.a .pioenvs/nodemcu/libGdbStub.a .pioenvs/nodemcu/libESP8266WiFi.a .pioenvs/nodemcu/libESP8266mDNS.a .pioenvs/nodemcu/libESP8266HTTPUpdateServer.a .pioenvs/nodemcu/libESP8266WebServer.a .pioenvs/nodemcu/libESP8266httpClient.a .pioenvs/nodemcu/libArduinoOTA.a .pioenvs/nodemcu/libSimpleTimer_ID419.a .pioenvs/nodemcu/libESPmanager.a .pioenvs/nodemcu/libFSBrowser.a -Wl,--end-group
/Users/amelvin/Arduino/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/nodemcu/firmware.elf section `.text' will not fit in region `iram1_0_seg'
```

bit stuck there... I essentially want to use platformio... which seems amazing, but seamlessly with the latest commits from https://github.com/esp8266/Arduino...  is this possible?

valeros, 2015-12-23:
Hi @sticilface 
Linking with ldscripts from https://github.com/esp8266/Arduino will not work.
Have you tried just place the latest framework to `.platformio\packages\framework-arduinoespressif` package?

sticilface, 2015-12-23:
Could you be more specific? by framework do you mean the latest git repo from ardunio... 

I refer to gihubrepo of arduino IDE - https://github.com/esp8266/Arduino  as ESParduindo
Placing this directory structure into `.platformio\packages\framework-arduinoespressif` does not work.. at all... edit:  if i add the version.txt it actually goes all the way through to the same error as i mentioned before .text iram

If I symbolically link (ln -s) in the SDK, cores, libraries folders  into `.platformio\packages\framework-arduinoespressif` then it compiles all the way, but fails due to the inclusion of TLS in the newer versions.. This has required different build flags etc... (bear with me, this is my first time venturing outside the arduino IDE ever.. I have 0 experience with the compilation process).  So i thought i could provide the linker scripts.. with a few modifications due to the fact that the paths are a bit different... now it fails with .text will not fit... seems like i'm almost almost there with it.. just not quite...

sticilface, 2015-12-23:
ok, that iram error is if i use the ardunio linker scripts.  If i revert and use the linker scripts from platformio i get this error
```
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/firmware.elf -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -Wl,-T"/Users/amelvin/.platformio/packages/ldscripts/esp8266.flash.4m.ld" .pioenvs/nodemcu/src/Melvana.o .pioenvs/nodemcu/src/Melvanimate.o .pioenvs/nodemcu/src/Melvtrix.o .pioenvs/nodemcu/src/Palette.o .pioenvs/nodemcu/src/_E131.o .pioenvs/nodemcu/src/effectobject.o .pioenvs/nodemcu/src/tmp_ino_to.o -L/Users/amelvin/.platformio/packages/framework-arduinoespressif/tools/sdk/lib -L/Users/amelvin/.platformio/packages/ldscripts -L.pioenvs/nodemcu -Wl,--start-group -lsmartconfig -lpp -lmain -lwpa -llwip -lnet80211 -lwps -lcrypto -lphy -lhal -laxtls -lgcc -lm .pioenvs/nodemcu/libFrameworkArduinoVariant.a .pioenvs/nodemcu/libFrameworkArduino.a .pioenvs/nodemcu/libNeoPixelBus.a .pioenvs/nodemcu/libAdafruit_GFX_ID13.a .pioenvs/nodemcu/libJson_ID64.a .pioenvs/nodemcu/libGdbStub.a .pioenvs/nodemcu/libESP8266WiFi.a .pioenvs/nodemcu/libESP8266mDNS.a .pioenvs/nodemcu/libESP8266HTTPUpdateServer.a .pioenvs/nodemcu/libESP8266WebServer.a .pioenvs/nodemcu/libESP8266httpClient.a .pioenvs/nodemcu/libArduinoOTA.a .pioenvs/nodemcu/libSimpleTimer_ID419.a .pioenvs/nodemcu/libESPmanager.a .pioenvs/nodemcu/libFSBrowser.a -Wl,--end-group
/Users/amelvin/.platformio/packages/framework-arduinoespressif/tools/sdk/lib/libcrypto.a(aes-internal.o):(.irom0.text+0x0): undefined reference to `rcons'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/nodemcu/firmware.elf] Error 1
========================= [ ERROR ] Took 19.47 seconds =========================
```

ivankravets, 2015-12-23:
@sticilface Sorry, I can't understand you :disappointed: Are you going to use latest Arduino Framework for ESP8266 or latest native SDK? This issue is related to native SDK.

sticilface, 2015-12-23:
ah!  doh!  my bad!  yes the arduino SDK :( sorry.... should i make a new post?

ivankravets, 2015-12-23:
@sticilface yes, please open separate issue

NoumanSaleem, 2016-02-28:
Is there a routine to update the native SDK? 1.5.2 is the latest provided by Espressif.

ctrowat, 2016-03-28:
I am also curious if there is any plan to update to the latest SDK?  I am working on porting a project over to use platformio but it requires SDK 1.5.1 or greater.
I did manually replace the contents of the .platformio/packages/sdk-esp8266 directory but that just doesn't feel like the right solution

ivankravets, 2016-03-28:
Guys, please give us a little time and we will resolve this issue.

ctrowat, 2016-03-28:
Excellent - thanks!

seclorum, 2016-04-06:
Just wanted to add a thumbs to updating to the 1.5.2 SDK, and the reason for that is that there have been plenty of bug-fixes and they're impacting my project progress, so .. no rush, just thumbs up for the hard work guys!

valeros, 2016-04-07:
I've just pushed some changes to the `espressif` platform, so If someone wants to test new SDK (v1.5.2) you need to update SDK package via `platformio update` and use the [development version](http://docs.platformio.org/en/latest/installation.html#development-version) of PlatformIO. Don't forget to specify your linker script in `platformio.ini` or use default from SDK package like in our [example](https://github.com/platformio/platformio/blob/develop/examples/espressif/esp8266-native/platformio.ini).

valeros, 2016-04-18:
If you have any problems with this issue please reopen it.

closed 2016-04-18 by valeros