platformio/platformio-core #367: Add : New ESP-01 (1M Flash)

itsjustvenky, 2015-12-10:
The new Arduino OTA (Arduino ESP8266) won't work on 512K (ESP-01). We need atleast 1M flash (SPIFFS choice based on your sketch size). 

TIP : We need to estimate the correct SPIFFS size, else ArduinoOTA won't work.

I had create new esp01 board inside "espressif.json", copy new "ld" for 1M flash from arduino-esp8266 and finally replaced whole 1.2 sdk with 1.4 sdk (from esp-open-sdk).

I just finished testing the Arduino OTA and it worked fine.

If it is possible for you to add a new 1M board to boards manager. thanks.

ivankravets, 2015-12-10:
> I had create new esp01 board inside "espressif.json"

Please don't modify @PlatformIO source code. You can create own board in the home directory. See http://docs.platformio.org/en/latest/platforms/creating_board.html

> copy new "ld" for 1M flash from arduino-esp8266 

Please run `platformio update`. The new `esp8266.flash.1m.ld` has been added to `~/.platformio/packages/ldscripts/`.

> finally replaced whole 1.2 sdk with 1.4 sdk (from esp-open-sdk).

@PlatformIO doesn't provide experimental software. We propose people only stable packages. However, you can use alpha/beta/gamma on your own risk. See stable SDK for Arduino/ESP8266 here https://github.com/esp8266/Arduino/tree/master/tools/sdk

itsjustvenky, 2015-12-10:
@ivankravets  Thank you so much. 

Now I understood why the SDK was still at v1.2. I will clean everything and get back to how it was before.

Appreciated!!

closed 2015-12-10 by itsjustvenky

itsjustvenky, 2015-12-10:
I confirm all four ld scripts for 1M ESP-01 board has been added.

ivankravets, 2015-12-10:
Thank you too that report it and help other users with @PlatformIO issues.