platformio/platformio-core #369: Serial monitor missing options

orome, 2015-12-11:
It seems some options found in some parts of the documentation for the serial monitor are missing, notably `format` and `echo`. I understand the latter isn't currently supported. Is there any chance of either repairing in future?

ivankravets, 2015-12-11:
The options for `platformio serialports monitor` depend on @PySerial version. See difference:
`PySerial 2.0`
```
platformio serialports monitor -h
Usage: platformio serialports monitor [OPTIONS]

Options:
  -p, --port TEXT       Port, a number or a device name
  -b, --baud INTEGER    Set baud rate, default=9600
  --parity [N|E|O|S|M]  Set parity, default=N
  --rtscts              Enable RTS/CTS flow control, default=Off
  --xonxoff             Enable software flow control, default=Off
  --rts [0|1]           Set initial RTS line state, default=0
  --dtr [0|1]           Set initial DTR line state, default=0
  --echo                Enable local echo, default=Off
  --cr                  Do not send CR+LF, send CR only, default=Off
  --lf                  Do not send CR+LF, send LF only, default=Off
  -d, --debug           Debug received data (escape non-printable chars)
                        #
                        --debug can be given multiple times:
                        # 0: just print
                        what is received
                        # 1: escape non-printable characters,
                        do newlines as unusual
                        # 2: escape non-printable
                        characters, newlines too
                        # 3: hex dump everything
  --exit-char INTEGER   ASCII code of special character that is used to exit
                        the application, default=29 (DEC)
  --menu-char INTEGER   ASCII code of special character that is used to
                        control miniterm (menu), default=20 (DEC)
  --quiet               Diagnostics: suppress non-error messages, default=Off
  -h, --help            Show this message and exit.
```

`PySerial 3.0`
```
platformio serialports monitor -h
Usage: platformio serialports monitor [OPTIONS]

Options:
  -p, --port TEXT       Port, a number or a device name
  -b, --baud INTEGER    Set baud rate, default=9600
  --parity [N|E|O|S|M]  Set parity, default=N
  --rtscts              Enable RTS/CTS flow control, default=Off
  --xonxoff             Enable software flow control, default=Off
  --rts [0|1]           Set initial RTS line state
  --dtr [0|1]           Set initial DTR line state
  --encoding TEXT       Set the encoding for the serial port (e.g. hexlify,
                        Latin1, UTF-8), default: UTF-8
  -f, --filter TEXT     Add text transformation
  --eol [CR|LF|CRLF]    End of line mode, default=CRLF
  --raw                 Do not apply any encodings/transformations
  --exit-char INTEGER   ASCII code of special character that is used to exit
                        the application, default=29 (DEC)
  --menu-char INTEGER   ASCII code of special character that is used to
                        control miniterm (menu), default=20 (DEC)
  --quiet               Diagnostics: suppress non-error messages, default=Off
  -h, --help            Show this message and exit.
```

ivankravets, 2015-12-11:
@PySerial 3.0 has not been released yet. However, you can try beta version
```
pip uninstall pyserial
pip install https://github.com/pyserial/pyserial/archive/master.zip

# revert back to PySerial 2.0
pip uninstall pyserial
pip install pyserial
```

orome, 2015-12-11:
@ivankravets — I have 2.7, but don't have `echo`.

ivankravets, 2015-12-11:
Give me please an output of 
```
platformio serialports monitor -h
``

orome, 2015-12-12:
I see:
```
Usage: platformio serialports monitor [OPTIONS]

Options:
  -p, --port TEXT       Port, a number or a device name
  -b, --baud INTEGER    Set baud rate, default=9600
  --parity [N|E|O|S|M]  Set parity, default=N
  --rtscts              Enable RTS/CTS flow control, default=Off
  --xonxoff             Enable software flow control, default=Off
  --rts [0|1]           Set initial RTS line state, default=0
  --dtr [0|1]           Set initial DTR line state, default=0
  --echo                Enable local echo, default=Off
  --cr                  Do not send CR+LF, send CR only, default=Off
  --lf                  Do not send CR+LF, send LF only, default=Off
  -d, --debug           Debug received data (escape non-printable chars)
                        #
                        --debug can be given multiple times:
                        # 0: just print
                        what is received
                        # 1: escape non-printable characters,
                        do newlines as unusual
                        # 2: escape non-printable
                        characters, newlines too
                        # 3: hex dump everything
  --exit-char INTEGER   ASCII code of special character that is used to exit
                        the application, default=29 (DEC)
  --menu-char INTEGER   ASCII code of special character that is used to
                        control miniterm (menu), default=20 (DEC)
  --quiet               Diagnostics: suppress non-error messages, default=Off
  -h, --help            Show this message and exit.
```
But `echo` has no effect.

ivankravets, 2015-12-12:
Try directly [python -m serial.tools.miniterm](https://pythonhosted.org/pyserial/examples.html#miniterm). Does it work for you?

orome, 2015-12-12:
Works the same. No echo.

ivankravets, 2015-12-12:
As I've mentioned, PlatformIO uses @PySerial for "serialports" command. See wrapping code https://github.com/platformio/platformio/blob/develop/platformio/commands/serialports.py#L149

How about to try PySerial 3.0?

ivankravets, 2015-12-14:
@orome please reopen this issue if still need help.

orome, 2015-12-14:
2.7 is the latest I can find.

ivankravets, 2015-12-14:
@orome see https://github.com/platformio/platformio/issues/369#issuecomment-164069177

closed 2015-12-16 by ivankravets