platformio/platformio-core #371: IOError: [Errno 2] No such file or directory: 'README.rst'

Macadamio, 2015-12-13:
hi i am trying to install on win7 and this error mesegge appear, i have try with:
pip -U install platformio and pip -U --egg install platformio and always apear this.

the last part of pip.log file is:

 Downloading from URL https://pypi.python.org/packages/source/p/platformio/platformio-2.5.0.tar.gz#md5=424a125f0899579cde8fd0456f1834aa (from https://pypi.python.org/simple/platformio/)
  Running setup.py (path:c:\users\carlos\appdata\local\temp\pip_build_Carlos\platformio\setup.py) egg_info for package platformio
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "c:\users\carlos\appdata\local\temp\pip_build_Carlos\platformio\setup.py", line 40, in <module>
    IOError: [Errno 2] No such file or directory: 'README.rst'
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "c:\users\carlos\appdata\local\temp\pip_build_Carlos\platformio\setup.py", line 40, in <module>

IOError: [Errno 2] No such file or directory: 'README.rst'

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in c:\users\carlos\appdata\local\temp\pip_build_Carlos\platformio
Exception information:
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\pip\basecommand.py", line 122, in main
    status = self.run(options, args)
  File "C:\Python27\lib\site-packages\pip\commands\install.py", line 278, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "C:\Python27\lib\site-packages\pip\req.py", line 1229, in prepare_files
    req_to_install.run_egg_info()
  File "C:\Python27\lib\site-packages\pip\req.py", line 325, in run_egg_info
    command_desc='python setup.py egg_info')
  File "C:\Python27\lib\site-packages\pip\util.py", line 697, in call_subprocess
    % (command_desc, proc.returncode, cwd))
InstallationError: Command python setup.py egg_info failed with error code 1 in c:\users\carlos\appdata\local\temp\pip_build_Carlos\platformio

ivankravets, 2015-12-13:
> File "", line 17, in
> File "", line 17, in

Try to upgrade setuptools before using `pip install -U setuptools pip`

Regards, Ivan Kravets
  - Ph.D, Researcher and Software Architect
  - "Creativity comes from talent and never from knowledge" (c)
________________________
http://www.ikravets.com

Macadamio, 2015-12-14:
hi 
i have upgrade setuptools and this solved the problem.
thanks!!!!

commit 94bb0dc4 (2015-12-14): Revert back step with upgrading for pip/setuptools // Resolve #371
closed 2015-12-14 by ivankravets