platformio/platformio-core #375: section `.text` will not fit in region `iram1_0_seg`

andriyadi, 2015-12-15:
I kept getting this error
```section '.text' will not fit in region 'iram1_0_seg'```

while building project for esp12e

Using the exact source code and build with Arduino IDE didn't raise that error. Please advice.

andriyadi, 2015-12-15:
FYI. I've updated platformio to the latest version.

Used libraries:
ESP8266WiFi
DNSServer
ESP8266WebServer
EEPROM
Wire

valeros, 2015-12-15:
Hi @andriyadi 
Can you share your project for testing?

andriyadi, 2015-12-15:
@valeros can I share to you privately, as this is my company's project. I'm afraid I can't share the project openly. Please share your email. Thanks.

valeros, 2015-12-15:
@andriyadi Sure, `valeros08@gmail.com`

andriyadi, 2015-12-15:
It's sent. Thanks in advance.

andriyadi, 2015-12-15:
Any workaround so far?

ivankravets, 2015-12-15:
Yes, this is a bug. I'm working on it. Please wait for the new develop release today.

ivankravets, 2015-12-15:
Please run `platformio update`. Does it work?

ivankravets, 2015-12-15:
Please upgrade to https://github.com/platformio/platformio/releases/tag/v2.6.0

closed 2015-12-15 by ivankravets

andriyadi, 2015-12-15:
It works flawlessly!

Thanks!

ivankravets, 2015-12-15:
Thanks you too that report this `typo` issue.