platformio/platformio-core #377: .text not fit in iram1_0_seg for esp12e, where do ldscripts live?

mangelajo, 2015-12-16:
This started happening after the last ldscripts update. Where can I find those?

/Users/ajo/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pioenvs/esp12e/firmware.elf section `.text' will not fit in region `iram1_0_seg'

It happens with this project:

https://github.com/ohaut/ray/tree/master/firmware/ray

It didn't happen before the ldscripts update.

valeros, 2015-12-16:
@mangelajo 
Please update packages via `platformio update`

ivankravets, 2015-12-16:
You can check version via `platformio --version`. Please use `pip install -U platformio` command in future.

As for your issue, could you share here or to me@ikravets.com project which causes this error? Thanks!

ivankravets, 2015-12-16:
@mangelajo I see that use duplicate PubSubClient library here https://github.com/ohaut/ray/tree/master/firmware/ray/lib/PubSubClient

I propose you to use [PlatformIO Library Manager](http://docs.platformio.org/en/latest/librarymanager/index.html) where this library is listed http://platformio.org/#!/lib/show/89/PubSubClient

You can add dependent libraries to your project using [lib_install](http://docs.platformio.org/en/latest/projectconf.html#lib-install)
```ini
[platformio]
src_dir = ./ray

[env:esp12e]
platform = espressif
framework = arduino
board = esp12e
lib_install = 89
```

The library will be automatically downloaded and compiled.

mangelajo, 2015-12-17:
Hmm @ivankravets, that's very nice, please keep reading for some suggestions.

 I'm always afraid that when somebody is going to recompile my software it will find compatibility issues with external libraries.

I have a suggestion here, may be I can open a separate bug:

What if we were able to define in platformio.ini the external libraries by name or some "human uuid" (pub-sub-client for example), and, important point here... (optionally) specific versions of such libraries?, this way when you try to compile the software, if those libraries are not downloaded, they get downloaded and installed, if there is any issue across version, you can fix it on a specific one, or a range (pip requirements style).

Guys you're doing an amazing job here!

mangelajo, 2015-12-17:
@valeros:

Updating ldscripts package:
Versions: Current=15, Latest=17 	 [Out-of-date]

Where's that stored, do we have a git for it?

It would be nice to track changes via git.

mangelajo, 2015-12-17:
@valeros, new version compiles well, what was the change?

itsjustvenky, 2015-12-17:
@mangelajo  you can always check the git commit for what is changed. always look for the corresponding branch.

ivankravets, 2015-12-17:
> Updating ldscripts package:
Versions: Current=15, Latest=17 [Out-of-date]
Where's that stored, do we have a git for it?
It would be nice to track changes via git.

See related issue #379. I will move LD scripts to repo.

ivankravets, 2015-12-17:
> What if we were able to define in platformio.ini the external libraries by name or some "human uuid" (pub-sub-client for example), and, important point here... (optionally) specific versions of such [...]

PlatformIO Library Manager worked in this manner 1 year ago. Then I had problems with TradeMarks & Copyrights. We can not use direct name for library. Yes, I understand that many developers use the TM in own libraries, for example `Arduino(tm)`. Developers don't understand that then can't do it, because Arduino(tm) doesn't allow it. See https://www.arduino.cc/en/Trademark/HomePage

> The product can't contain the name Arduino
>  If you sell a sensor that is compatible with Arduino you can call it "XYZ sensor for Arduino" and NOT "Arduino XYZ Sensor"

In this case the correct name for library "ArduinoBob" is "Bob for Arduino". 

-----
Summary. Then I totally rewrote logic of PlatformIO Library Manager where users don't operate with trademarks, they have unique ID of the library. Here are more advantages, but also are disadvantage, like "need to know the ID of LIB". Please use our http://platformio.org/#!/lib or http://docs.platformio.org/en/latest/librarymanager/index.html

PlatformIO is non-commercial project, free and open source. I don't have time/money to waste on these TM monsters.

ivankravets, 2015-12-17:
> new version compiles well, what was the change?

It was my mistake. I omitted this line `*libm.a:(.literal .text .literal.* .text.*)` for `.irom0.text : ALIGN(4)` section.

ivankravets, 2015-12-17:
Please reopen this issue if you have any questions

closed 2015-12-17 by ivankravets

ivankravets, 2015-12-18:
@mangelajo @itsjustvenky PlatformIO 2.6.1 is released! Please upgrade https://github.com/platformio/platformio/releases/tag/v2.6.1