platformio/platformio-core #378: Please add esp12 board with 3M Sketch, 1M Spiff

itsjustvenky, 2015-12-16:
Currently the file '~/.platformio/packages/ldscripts/esp8266.flash.4m.ld" has
1MB Sketch 
3MB spiffs.

```
/* Flash Split for 4M chips */
/* sketch 1019KB */
/* spiffs 3052KB */
/* eeprom 20KB */
```

In Arduino/esp8266 package there is another option (eagle.flash.4m1m.ld) with 
1MB SPIFFS 
1MB Sketch 
2MB free/empty

```
/* Flash Split for 4M chips */
/* sketch 1019KB */
/* empty  2048KB */
/* spiffs 1004KB */
/* eeprom 20KB */
```

Can you please add this to boards.Thanks

ivankravets, 2015-12-16:
Added. Please `platformio update`. See
```
➜  ~  ls -l ~/.platformio/packages/ldscripts/esp8266.flash.*
-rwxrwxr-x  1 ikravets  staff  20766 Dec 16 18:27 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.16m.ld
-rwxrwx---  1 ikravets  staff  20641 Dec 15 18:36 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.1m128.ld
-rwxrwx---  1 ikravets  staff  20641 Dec 15 18:36 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.1m256.ld
-rwxrwx---  1 ikravets  staff  20640 Dec 15 18:36 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.1m512.ld
-rwxrwx---  1 ikravets  staff  20640 Dec 15 18:36 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.1m64.ld
-rwxrwxr-x  1 ikravets  staff  20767 Dec 16 18:27 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.2m.ld
-rwxrwx---  1 ikravets  staff  20769 Dec 16 11:02 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.4m.ld
-rwxrwxr-x  1 ikravets  staff  20787 Dec 16 18:27 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.4m1m.ld
-rwxrwx---  1 ikravets  staff  20767 Dec 15 18:34 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.512k.ld
-rwxrwxr-x  1 ikravets  staff  20750 Dec 16 18:27 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.512k0.ld
-rwxrwxr-x  1 ikravets  staff  20767 Dec 16 18:27 /Users/ikravets/.platformio/packages/ldscripts/esp8266.flash.8m.ld
```

How to use specific LD script? http://docs.platformio.org/en/latest/projectconf.html#build-flags
Example,
```ini
[env:specific_ld_script]
build_flags = -Wl,-T~/.platformio/packages/ldscripts/esp8266.flash.512k0.ld
```

ivankravets, 2015-12-16:
We will improve handling of built-in LD scripts. See https://github.com/platformio/platformio/issues/379

closed 2015-12-16 by ivankravets

itsjustvenky, 2015-12-16:
@ivankravets  thank you.