platformio/platformio-core #380: esp12e OSX fails to upload a basic example

aperepel, 2015-12-16:
Hi, trying out the most basic blinky example with PlatformIO. Everything works fine with ArduinoIDE as well as Eclipse Arduino. I, however, am running into issues with PlatformIO (the code is the most basic blink/delay):

```
platformio run --target upload
[Wed Dec 16 15:43:50 2015] Processing esp12e (platform: espressif, board: esp12e, framework: arduino)
-----------------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/esp12e/firmware.bin"])
Auto-detected UPLOAD_PORT/DISK: /dev/cu.wchusbserial1410
/Users/agrande/.platformio/packages/tool-esptool/esptool -vv -cd ck -cb 115200 -cp /dev/cu.wchusbserial1410 -ca 0x00000 -cf .pioenvs/esp12e/firmware.bin
esptool v0.4.5 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to ck
setting baudrate from 115200 to 115200
setting port from /dev/tty.usbserial to /dev/cu.wchusbserial1410
setting address from 0x00000000 to 0x00000000
espcomm_upload_file
stat .pioenvs/esp12e/firmware.bin success
opening port /dev/cu.wchusbserial1410 at 115200
tcgetattr
tcsetattr
serial open
opening bootloader
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
scons: *** [upload] Error 255
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
resetting board
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
trying to connect
setting character timeout 0
done
setting character timeout 1
done
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
============================================= [ ERROR ] Took 3.11 seconds =============================================
```

ivankravets, 2015-12-16:
Could you provide the same output from Arduino IDE/Eclipse IDE? I mean this `/Users/agrande/.platformio/packages/tool-esptool/esptool -vv -cd ck -cb 115200 -cp /dev/cu.wchusbserial1410 -ca 0x00000 -cf .pioenvs/esp12e/firmware.bin`

Thanks.

ivankravets, 2015-12-16:
Please re-test it with the latest http://docs.platformio.org/en/latest/installation.html#d-development-version

closed 2015-12-16 by ivankravets

ivankravets, 2015-12-16:
Don't forget to `platformio update` all packages

aperepel, 2015-12-16:
Thanks! This worked. Appreciate a prompt fix.

ivankravets, 2015-12-16:
Which is type of your board? NodeMCU?

aperepel, 2015-12-17:
Yes, that's the one.

ivankravets, 2015-12-18:
@aperepel PlatformIO 2.6.1 is released! Please upgrade https://github.com/platformio/platformio/releases/tag/v2.6.1

zdebord, 2016-08-17:
Same problem for the Expressif ESP-01:

espcomm_send_command: sending command header
espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed

commit 34a5d366 (2015-12-16): Fix reset method for Espressif NodeMCU (ESP-12E Module) // Resolve #380