platformio/platformio-core #383: Eclipse : duplicate path entries found in project path

itsjustvenky, 2015-12-17:
I am latest version 

```
PlatformIO, version 2.6.0
```
and libraries, tools are updated using "platformio update"

Now I have created a eclipse project

```
platformio init --ide eclipse --board esp12e
```

and I imported the project into eclipse (mars) using Import >> Existing projects into WorkSpace
and when I build the project I see the following warning in "Problems" tab.

```
Description	Resource	Path	Location	Type
Invalid project path: Duplicate path entries found (/esp12e_sleep [Include path] isSystemInclude:true includePath:/home/venkat/.platformio/packages/toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/include), path: [/esp12e_sleep].	esp12e_sleep		pathentry	Path Entry Problem
```

itsjustvenky, 2015-12-17:
There is no issue apart from the warning, code compiles and uploads to ESP properly. I just want to get rid of that warning.

aperepel, 2015-12-17:
I can confirm I saw that too, something simple is definitely going on.

ivankravets, 2015-12-17:
Guys, help us. Please go to project settings and investigate which INCLUDES are unnecessary.

* Eclipse templates: https://github.com/platformio/platformio/tree/develop/platformio/ide/tpls/eclipse
* IDE data for Project Generator: `platformio run -e YOUR_ENV -t idedata`

Here is my dump
```
platformio run -d examples/espressif/esp8266-webserver -e esp12e -t idedata
[Thu Dec 17 16:24:21 2015] Processing esp12e (platform: espressif, board: esp12e, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
{"includes": ["/Users/ikravets/.platformio/packages/framework-arduinoespressif/variants/nodemcu", "/Users/ikravets/.platformio/packages/framework-arduinoespressif/cores/esp8266", "/Users/ikravets/.platformio/packages/framework-arduinoespressif/cores/esp8266/spiffs", "/Users/ikravets/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src", "/Users/ikravets/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/arch", "/Users/ikravets/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/include", "/Users/ikravets/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/lwip", "/Users/ikravets/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WebServer/src", "/Users/ikravets/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WebServer/src/detail", "/Users/ikravets/.platformio/packages/framework-arduinoespressif/libraries/ESP8266mDNS", "/Users/ikravets/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include", "/Users/ikravets/.platformio/packages/toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/include", "/Users/ikravets/.platformio/packages/toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/include-fixed"], "cxx_path": "/Users/ikravets/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++", "defines": ["F_CPU=80000000L", "__ets__", "ICACHE_FLASH", "ARDUINO_ESP8266_ESP12", "ARDUINO_ARCH_ESP8266", "ESP8266", "ARDUINO=200", "PLATFORMIO=020601"]}
```

JSON formatter: https://jsonformatter.curiousconcept.com

ivankravets, 2015-12-17:
@aperepel please say thanks via :star: to @valeros. He likes them :blush: Thanks.

itsjustvenky, 2015-12-17:
```
{
	"includes": ["/home/venkat/.platformio/packages/framework-arduinoespressif/variants/nodemcu",
	"/home/venkat/.platformio/packages/framework-arduinoespressif/cores/esp8266",
	"/home/venkat/.platformio/packages/framework-arduinoespressif/cores/esp8266/spiffs",
	"/home/venkat/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src",
	"/home/venkat/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/arch",
	"/home/venkat/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/include",
	"/home/venkat/.platformio/packages/framework-arduinoespressif/libraries/ESP8266WiFi/src/lwip",
	"/home/venkat/.platformio/packages/framework-arduinoespressif/libraries/ESP8266mDNS",
	"/home/venkat/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include",
	"/home/venkat/.platformio/packages/toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/include",
	"/home/venkat/.platformio/packages/toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/include-fixed"],
	"cxx_path": "/home/venkat/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++",
	"defines": ["F_CPU=80000000L",
	"__ets__",
	"ICACHE_FLASH",
	"ARDUINO_ESP8266_ESP12",
	"ARDUINO_ARCH_ESP8266",
	"ESP8266",
	"ARDUINO=200",
	"PLATFORMIO=020600"]
```

ivankravets, 2015-12-17:
@itsjustvenky I don't see duplicated entries your list. I'll test it later.

commit 7119bb90 (2015-12-18): Fix project generator for Eclipse IDE and "duplicate path entries found in project path" // Resolve #383
closed 2015-12-18 by ivankravets