platformio/platformio-core #384: Monitoring a serial port breaks upload

aperepel, 2015-12-17:
Board: esp12e

If I attach to a serial port to monitor the output, it will prevent upload from working. Not sure what can be done here, but other environments manage to somehow do it. Reverse is true - if I start the upload process and attach to a serial port in another terminal window, it will fail the upload (and attach too).

Ideally, we'd like to keep the monitor open and attached and keep re-uploading the firmware iteratively. This way the monitor won't be skipping some init output and have us re-attach after every upload.

ivankravets, 2015-12-17:
> If I attach to a serial port to monitor the output, it will prevent upload from working. Not sure what can be done here

Nothing can be done here :worried: Device is busy by another process. The only one solution is terminate current process using administrative privileges. However, I don't recommend to use `platformio` with administrative/sudo permissions.

> but other environments manage to somehow do it

Do you mean Arduino IDE? They have PID of opened serial monitor and can close/kill it before uploading and then restore it. 

PlatformIO is full flexible and independent toolkit. We don't limit users in their actions. Someone likes to use external Serial Monitor with GUI, others like [platformio serialports monitor]  (http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#platformio-serialports-monitor). We don't force users to use all our tools from @PlatformIO. Our aim is to develop useful tools and improve obsolete practice with embedded development. User have ability to decide what is comfortable for him.

As for me, I use [CoolTerm](http://freeware.the-meiers.org). Of course, each time before uploading I manually disconnect/connect to SerialPort in GUI.

itsjustvenky, 2015-12-17:
The Sming framework does this similar to arduino IDE from command line when used "make flash".
After flashing, it open its own serial monitor

```
TERMINAL     ?= python -m serial.tools.miniterm $(COM_PORT) $(COM_SPEED_SERIAL)
```
```
flash: all
	$(vecho) "Killing Terminal to free $(COM_PORT)"
	-$(Q) $(KILL_TERM)
ifeq ($(DISABLE_SPIFFS), 1)
	$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(FW_BASE)/0x00000.bin 0x09000 $(FW_BASE)/0x09000.bin
else
	$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(FW_BASE)/0x00000.bin 0x09000 $(FW_BASE)/0x09000.bin $(SPIFF_START_OFFSET) $(SPIFF_BIN_OUT)
endif
	$(TERMINAL)
```

Sorry I don't want to mention other frameworks here, but just for example/reference purpose I providing the idea.

If it all this is possible to implement with a extra option to platformio command line, then it will be a great feature.

Thoughts ?

ivankravets, 2015-12-17:
I've explained above the problem. All these IDEs work in own isolated space where they spawn child process and have their PIDs. You can reproduce same logic using own "isolated space" via script `upload_and_monitor.sh`
```
platformio run -t upload
platformio serialports monitor
```

What is more, you can put these line into infinite loop. `Exit from Monitor` will raise uploading, and ... repeat.

aperepel, 2015-12-17:
I understand your point, Ivan. Was really checking if there was any trick or obvious omission, but was on a cautious side.

For my purposes this one works quite well:
```
platformio run -t upload && platformio serialports monitor --baud 115200
```
As soon as upload is done - it connects to the monitor, so I don't skip any early init output. Maybe we can put it in a FAQ as a best practice?

itsjustvenky, 2015-12-17:
@aperepel @ivankravets  thanks.

closed 2015-12-17 by ivankravets

XhstormR, 2019-11-30:
I use the following command and it is very useful:
```bash
pio run --target upload --target monitor
```

commit edc06f12 (2015-12-17): Add to FAQ "Monitoring a serial port breaks upload" // Resolve #384