platformio/platformio-core #386: Serial Monitor: Exit key combination doesn't work on international keyboards

amendoim, 2015-12-17:
In Windows the platformio serialports monitor Exit key combination Ctrl + ] does not work on international keyboards where ] is produced by a combination of keys.

For example in the standard Portuguese keyboard (Portugal and probably Brazil and other variants also), the ] character is produced by pressing Alt Gr + 8. In this case Serial Monitor never exits and one has to press Ctrl + Break to kill the program.

ivankravets, 2015-12-17:
You can override exit key. See http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#cmdoption-platformio-serialports-monitor--exit-char

amendoim, 2015-12-18:
That will send the exit char but won't quit the app itself, right? Can't platformio use an universal key combination like Ctrl + Q? 
I suspect many other non-us keyboards will have this behavior...

ivankravets, 2015-12-18:
> Can't platformio use an universal key combination like Ctrl + Q? 

Universal for what? No here "universal". You can configure own `--exit-char`. See
```shell
platformio serialports monitor --exit-char 93
--- Miniterm on /dev/cu.usbmodemFD1311: 9600,8,N,1 ---
--- Quit: ']'  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

# to exit use Ctrl+C+]

platformio serialports monitor --exit-char 17
--- Miniterm on /dev/cu.usbmodemFD1311: 9600,8,N,1 ---
--- Quit: Ctrl+Q  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

# to exit use Ctrl+Q
```

If you specify DIRECT ASCII code (like ``93`` for ``]``), then you need to use `Ctrl+C+]``. To use direct key, please minus 64 from ASCII char. For example, ASCII code for `'Q'` char is `81`, then `81-64=17`. Now `platformio serialports monitor --exit-char 17` will wait for `Ctrl+Q`

Here is the table http://www.theasciicode.com.ar/ascii-printable-characters/box-brackets-square-brackets-closing-bracket-ascii-code-93.html

Does it help you?

P.S: Please give us a star (:star:) https://github.com/platformio/platformio/stargazers

amendoim, 2015-12-19:
I meant universal for all keyboards: the default you have now will work on keyboards where Ctrl + ] is a direct key combination (like the English keyboard) in others it won't work (nothing happens).
Because of this maybe it would make sense to have a default exit combination that works for everyone, like Ctrl + Q or similar.

In any case, using --exit-char works fine. I'd suggest adding the explanation above to the docs as this information is far from obvious.

Thanks, you deserve many stars, this is an awesome project! Regards!

ivankravets, 2015-12-19:
Which `--exit-char` do you use? I'll add to the doc. 

Thanks!

amendoim, 2015-12-19:
Ctrl + Q as you listed two posts ago: --exit-char 17

ivankravets, 2015-12-22:
I've added to docs. What is interesting, that `Ctrl+Q` doesn't work for me :disappointed:

commit 943d16bc (2015-12-22): Add example with "Ctrl+Q" to serial ports monitor as exit key // Resolve #386
closed 2015-12-22 by ivankravets