platformio/platformio-core #387: Build fails when platformio trys to compile and the build path contains spaces.

Seep1959, 2015-12-17:
On windows platformio seems to by default keep toolchains, ldscrips, and librarys in "C:\Users\<Username>\\.platformio . When the username contains a space, platformio fails when attempting to compile do to the path being split at the space. [Here is my terminal output](https://gist.github.com/Seep1959/dc485620d0a5420076d3) when I run "platformio run".

On line 252 and 253 you can see

arm-none-eabi-gcc: error: C:\Users\Firstname: No such file or directory
arm-none-eabi-gcc: error: LastName\\.platformio\packages\framework-libopencm3\lib\stm32\f4\stm32f405x6.ld: No such file or directory

Instead of the path being  

>"C:\Users\Firstname Lastname\\.platformio\packages\framework-libopencm3\lib\stm32\f4\stm32f405x6.ld" 

it is split to become 

>"C:\Users\Firstname" and "Lastname\\.platformio\packages\framework-libopencm3\lib\stm32\f4\stm32f405x6.ld" 

causing it to fail building.

I am running on windows 8.1 64bit, platformio version 2.6.2, and python version 2.7.11

If anyone knows a temporary work around, possibly storing .platformio in a different directory. I would really appreciate it.

ivankravets, 2015-12-17:
I'll look on it. Temporary solution http://docs.platformio.org/en/latest/faq.html#windows-unicodedecodeerror-ascii-codec-can-t-decode-byte

Regards, Ivan Kravets
  - Ph.D, Researcher and Software Architect
  - "Creativity comes from talent and never from knowledge" (c)
________________________
http://www.ikravets.com

closed 2015-12-18 by ivankravets

ivankravets, 2015-12-18:
@Seep1959 PlatformIO 2.6.1 is released! Please upgrade https://github.com/platformio/platformio/releases/tag/v2.6.1

commit 186580d7 (2015-12-18): Fix issue with code builder when build path contains spaces // Resolve #387
commit 0951d53f (2015-12-18): Fix issue with code builder when build path contains spaces // Resolve #387