platformio/platformio-core #388: Undefined reference to main error. collect2.exe: error: ld returned 1 exit status

Seep1959, 2015-12-18:
I am trying to compile the stm32f4discovery-libopencm3 blink example and I am getting this error, [here is the terminal log](https://gist.github.com/Seep1959/66d92eedf24e8edb0086)

valeros, 2015-12-18:
Hi @Seep1959 
Have you implemented `main` function?
I compiled this `miniblink` https://github.com/libopencm3/libopencm3-examples/blob/master/examples/stm32/f4/stm32f4-discovery/miniblink/miniblink.c without any errors.

Seep1959, 2015-12-18:
Yes, I did. I have actually tried both the example you linked and the platformio example.

ivankravets, 2015-12-18:
@Seep1959 please give us a star (:star:) https://github.com/platformio/platformio/stargazers for the help

valeros, 2015-12-18:
I don't see any others files built except framework files.
Could you provide your project structure?

Seep1959, 2015-12-18:
Okay, well I feel like a bit of an idiot now. Turns out my c file was saved as a txt file. I saved it as a proper c file and it compiled correctly.

ivankravets, 2015-12-18:
Cool. Have a nice coding and weekend! :beer:  :smile:

closed 2015-12-18 by ivankravets