platformio/platformio-core #389: Using PlatformIO with Atom || Changing between .ino files

tiferrei, 2015-12-18:
Hi, I am unsure if I should ask this here or in an atom forum, but I have Platomforino installed on Atom, running perfectly, the only thing I am intrigued is if I have to init a PlatformIO project for each .ino file that I want to upload. Isn't there a way of PlatformIO only compile and Upload the file I have open on Atom?

Thank you,
Tiago Ferreira

[image]

ivankravets, 2015-12-18:
HI, please create 2 separate projects. PlatformIO doesn't process only SINGLE opened file, it processes all files from `src` directory. However, it uses opened file as "indicator" where need to find `src` directory.

tiferrei, 2015-12-18:
Actually I came up with a possible solution, join all the programs as functions and just call the specific function on the void loop()

ivankravets, 2015-12-18:
Do you need additional help?

ivankravets, 2015-12-18:
Don't forget to give us a star :star:  https://github.com/platformio/platformio/stargazers

Thanks!

tiferrei, 2015-12-18:
Nope! By the way I think you made a great job, I really like the idea of PlatformIO, give me any link for reviews I can give, you got another fan! ;)

ivankravets, 2015-12-18:
Thanks for :star: https://github.com/platformio/platformio/stargazers ! Here is a list with articles about us http://docs.platformio.org/en/latest/articles.html

Don't hesitate to contact us if have any questions.

tiferrei, 2015-12-18:
Thanks! Keep up the great work!

closed 2015-12-18 by tiferrei

ivankravets, 2015-12-18:
Where is star :star:  for us? :blush:

tiferrei, 2015-12-18:
How do I "star"?

ivankravets, 2015-12-18:
Thanks! It helps grow up our project!

tiferrei, 2015-12-18:
Wish you best of luck and a Merry Christmas!

ivankravets, 2015-12-18:
Thanks! :smile: You too! Happy Holidays! :christmas_tree: