platformio/platformio-core #390: 2.6.1 install generates bad esp12e firmware

aperepel, 2015-12-18:
Something has happened and the simplest possible code never proceeds beyond `setup()`. It's just stuck after executing (or never exiting?) the setup() call. I've tried the same code in Arduino IDE 1.6.5 and things work ok.

```
// a function to be executed periodically
void repeatMe() {
  Serial.print("Uptime (s): ");
  Serial.println(millis() / 1000);
}

void setup() {
  Serial.begin(9600);
  Serial.print("In setup()");
}

void loop() {
  Serial.println("In loop.");
  repeatMe();
  delay(1000);
}
```

ivankravets, 2015-12-18:
Which type of board do you use in Arduino IDE? Flash Size? CPU Frequency?

aperepel, 2015-12-18:
ESP12-E
80MHz
Flash 4M (3M SPIFFS)
115200 upload speed

ivankravets, 2015-12-18:
Do you use `nodemcu` type of board?

aperepel, 2015-12-18:
Yes, it is technically a nodemcu dev board (but I don't use the lua env).

This whole story didn't sit well with me, breaking out of the blue (must have picked up an update and missed the message). I decided to wipe the `~/.platformio` folder (pip uninstall didn't want to do it for me), redownloaded the `espressif` platform and finally restored my sanity. I hope it was simply a fluke with my setup - jumping from 2.6.0 -> dev -> 2.6.1, the upgrade didn't work out :)

Thanks for prompt responses.

ivankravets, 2015-12-18:
Do you mean that it works now in the latest 2.6.1?

aperepel, 2015-12-18:
Yes, in a clean environment. I'm not sure if there is a problem with pip uninstall (I ran `sudo -H pip uninstall platformio`, but it didn't remove `~/.platformio`).

ivankravets, 2015-12-18:
It was problem with `ldscripts`. Looks like you didn't update them when upgraded to 2.6.1.

Ok, this is good news! We have a big interest from new users to Espressif platform. Need to polish it. 
Don't hesitate to open new issue if you will have any problems. Thanks!

closed 2015-12-18 by ivankravets