platformio/platformio-core #391: st-flash.exe app crash, scons: *** [upload] Error 309

Seep1959, 2015-12-18:
[Image of app crash](http://imgur.com/NwzMxne)

[Terminal log](https://gist.github.com/Seep1959/210daaf0e81f396cf8e1)

I have tried replacing the libusb dll with an updated one but then I just get error 123, so I reverted back to the original libusb. I am able to successfully program the stm32f4discovery with the .bin platformio generates using the stm32 ST-LINK Utility. Any idea why platformIOs st-flash tool is crashing?

valeros, 2015-12-21:
Hi @Seep1959 
What is wrong with default `libusb-1.0.dll`?

Seep1959, 2015-12-22:
Nothing what wrong with it, I thought it might fix it.

valeros, 2015-12-24:
@Seep1959 Please, update your `st-link` package via `platformio update`

ivankravets, 2015-12-24:
@Seep1959 please reopen this issue if it doesn't work.

closed 2015-12-24 by ivankravets