platformio/platformio-core #392: Arduino & Engduino Serial Monitor

tiferrei, 2015-12-19:
Hi, sorry if this is a stupid question but how can I see the `<Serial.println>` output?

Thanks,

Tiago Ferreira

[image]

ivankravets, 2015-12-19:
Platomformio Package for Atom editor doesn't have built in serial monitor. Try to find related package in Atom packages registry. 

Nevertheless, you can use native [platformio serialports monitor](http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#platformio-serialports-monitor).

ivankravets, 2015-12-21:
If you have any questions please reopen this issue. Thanks.

closed 2015-12-21 by ivankravets