platformio/platformio-core #393: installation failed on linux-x86 (ubunut 15.04)

framlin, 2015-12-20:
Servus,

I tried to install platformio, but it fails with the following message:

Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-pLMzk4/scons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-EveMj1-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-pLMzk4/scons

[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio/issues >

Can anybody please give me a hint, what is going wrong?

framlin, 2015-12-20:
Sorry, I did not "install the udev-rules ....."

closed 2015-12-20 by framlin

ivankravets, 2015-12-20:
Does it work for you now?

ivankravets, 2015-12-20:
Would be grateful for the :star: https://github.com/platformio/platformio/stargazers

framlin, 2015-12-25:
yepp, it does work now

ivankravets, 2015-12-25:
Thanks for the report. Would be thankful for the :star: https://github.com/platformio/platformio/stargazers

Happy Holidays and Merry Christmas! :christmas_tree: