platformio/platformio-core #394: Digispark Micronucleus Upload: Not Found

peterkjacobson, 2015-12-20:
I am using a Digispark board and when I upload the script does not seem to be able to locate the micronucleus program.  I have verified that micronucleus does appear in the correct folder and I can upload manually, but PlatformIO doesn't seem to be able to use it.  The error is as follows:

[Sun Dec 20 12:09:15 2015] Processing digispark-tiny (platform: atmelavr, board: digispark-tiny, framework: arduino)
--------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/digispark-tiny/firmware.hex"])
Please unplug/plug device ...
"/home/peter/.platformio/packages/tool-micronucleus/micronucleus" -c digispark --timeout 60 .pioenvs/digispark-tiny/firmware.hex
sh: 1: /home/peter/.platformio/packages/tool-micronucleus/micronucleus: not found
scons: *** [upload] Error 127
========================= [ ERROR ] Took 0.56 seconds =========================
[Finished in 0.954s]

peterkjacobson, 2015-12-20:
Nevermind, somehow the folder was changed from "tool-micronucleus" to "tools-micronucleus."  My fault I imagine.

ivankravets, 2015-12-20:
I've just checked you board and could not repeat this issue. Did you renamed this folder manually?

ivankravets, 2015-12-20:
Hm... Could you remove totally `/home/peter/.platformio` folder and try build project from the scratch? 

This is important for us. Need to understand that all works good. Thanks.

peterkjacobson, 2015-12-21:
I will try this tonight and report back to you.  Thanks for the quick response!

ivankravets, 2015-12-21:
Thanks! I reassigned this issue to 2.6.3 release. Will wait for your report.

ivankravets, 2015-12-22:
Please reopen this issue if need help

peterkjacobson, 2015-12-23:
So I deleted .platformio and attempted to run, and the same issue actually appeared.  It installed micronucleus properly and I can see it in the right folder, but even if I go to that folder in terminal and run micronucleus directly it says "command not found."

Here is the output from platformio:

[Tue Dec 22 19:07:44 2015] Processing digispark-tiny (platform: atmelavr, board: digispark-tiny, framework: arduino)
--------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/digispark-tiny/firmware.hex"])
Please unplug/plug device ...
"/home/peter/.platformio/packages/tool-micronucleus/micronucleus" -c digispark --timeout 60 .pioenvs/digispark-tiny/firmware.hex
sh: 1: /home/peter/.platformio/packages/tool-micronucleus/micronucleus: not found
scons: *** [upload] Error 127
========================= [ ERROR ] Took 0.56 seconds =========================
[Finished in 0.967s]

peterkjacobson, 2015-12-23:
If I run ./micronucleus in the directory it does work.

ivankravets, 2015-12-23:
@peterkjacobson which is your host OS? Could you try this solution https://github.com/platformio/platformio/issues/400#issuecomment-166862334 ?

peterkjacobson, 2015-12-23:
> OS? Could you try this solution #400 (comment)

I will test when I get home tonight. Previously I had resolved this by
downloading and making the micronucleus file and putting it into the
folder.  I will confirm tonight.

peterkjacobson, 2015-12-24:
I am on 64 bit Ubuntu 15.10.  Replacing the micronucleus file with a 64 bit version fixed the problem and it uploads as expected.

valeros, 2015-12-24:
@peterkjacobson  Please, update your `tool-micronucleus` package via `platformio update`

ivankravets, 2015-12-24:
@peterkjacobson please reopen this issue if it doesn't work.

P.S: Please give us a :star: here https://github.com/platformio/platformio/stargazers :blush: :christmas_tree:

closed 2015-12-24 by ivankravets