platformio/platformio-core #395: Create package for Homebrew Mac OS X packages manager

ivankravets, 2015-12-20:
Subj.

ivankravets, 2015-12-30:
@uetchy do you have any problems with PlatformIO installation?

uetchy, 2015-12-30:
@ivankravets No. But it's slight issue around versioning of Python.
I'm using mainly Python 3 and I don't want to switch Python version to 2.x.

Homebrew can separately handle Python version required by packages. It means I can use platformio without switching Python.
That's why I need Homebrew support!

ivankravets, 2015-12-30:
@uetchy Could you create Homebrew's formula for `platformio`. The only one dependency should be `pip2`.

uetchy, 2015-12-30:
Actually, I already created [Homebrew Formula for platformio](https://gist.github.com/uetchy/e752ff8f8315a763b7b8) based on [official reference](https://github.com/Homebrew/homebrew/blob/master/share/doc/homebrew/Python-for-Formula-Authors.md).

But it failed with error:
```console
$ brew install platformio
==> Downloading https://github.com/platformio/platformio/archive/v2.6.3.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio-2.6.3.tar.gz
==> Downloading https://pypi.python.org/packages/source/b/bottle/bottle-0.12.9.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--bottle-0.12.9.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platfo
==> Downloading https://pypi.python.org/packages/source/c/click/click-5.1.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--click-5.1.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platfo
==> Downloading https://pypi.python.org/packages/source/c/colorama/colorama-0.3.5.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--colorama-0.3.5.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platfo
==> Downloading https://pypi.python.org/packages/source/l/lockfile/lockfile-0.12.2.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--lockfile-0.12.2.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platfo
==> Downloading https://pypi.python.org/packages/source/p/pyserial/pyserial-2.7.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--pyserial-2.7.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platfo
==> Downloading https://pypi.python.org/packages/source/r/requests/requests-2.9.1.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--requests-2.9.1.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platfo
==> Downloading https://pypi.python.org/packages/source/S/SCons/scons-2.4.1.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--SCons-2.4.1.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platfo
Last 15 lines from /Users/oame/Library/Logs/Homebrew/platformio/07.python:
__file__ = 'setup.py'
exec(compile(getattr(tokenize, 'open', open)(__file__).read()
  .replace('\r\n', '\n'), __file__, 'exec'))
--no-user-cfg
install
--prefix=/usr/local/Cellar/platformio/2.6.3/libexec/vendor
--single-version-externally-managed
--record=installed.txt

usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: -c --help [cmd1 cmd2 ...]
   or: -c --help-commands
   or: -c cmd --help

error: option --single-version-externally-managed not recognized

READ THIS: https://git.io/brew-troubleshooting
```

ivankravets, 2015-12-30:
1. Why do you specify all packages manually?
2. Could `platformio` be installed via `pip`? It uses it.

P.S: SCons was removed in PlatformIO 2.7.0, which will be released today.

uetchy, 2015-12-30:
1. `resource "something"` is necessary for [avoiding dependency collision](https://github.com/Homebrew/homebrew/blob/master/share/doc/homebrew/Python-for-Formula-Authors.md#dependencies) around `site-packages`.
2. Because of Homebrew policy. Homebrew Formula should build softwares from source.

I think this comment looks like the problem.
https://github.com/Homebrew/homebrew-python/issues/132#issuecomment-68668771

ivankravets, 2015-12-30:
Cool. Could you remove `scons` from requirement list and re-test?

ivankravets, 2016-02-27:
@uetchy Could you remove `SCons` from requirements list and update PlatformIO to the latest version?

uetchy, 2016-02-27:
Updated: https://gist.github.com/uetchy/e752ff8f8315a763b7b8
It looks work well :+1: 

You can test this Formula with:
```bash
$ brew install https://gist.githubusercontent.com/uetchy/e752ff8f8315a763b7b8/raw/b694a1668dae0c4654f7a24faaadf879c4ff4212/platformio.rb
```

ivankravets, 2016-02-27:
@uetchy  Hm...It works!!! Thanks!

Is difficult to make PR to the original Hombrew repo with our formula?

ivankravets, 2016-02-27:
@uetchy will you have any objection if I make PR to https://github.com/Homebrew/homebrew/pulls using your formula?

I hope that they will merge it from me.

Thanks.

uetchy, 2016-02-27:
@ivankravets Hooray! :raised_hands: 
I don't care about that. If anything, you ought to send PR.

ivankravets, 2016-02-29:
Thanks for all!!! Done https://twitter.com/platformio_org/status/704287776490004480

closed 2016-02-29 by ivankravets

ip2k, 2016-03-07:
When trying `brew install platformio`, I had to `pip install pbr` for it to work:

```
00:42 $ brew install platformio
==> Downloading https://homebrew.bintray.com/bottles/platformio-2.8.4.el_capitan.bottle.tar.gz

curl: (51) SSL: certificate verification failed (result: 5)
Error: Failed to download resource "platformio"
Download failed: https://homebrew.bintray.com/bottles/platformio-2.8.4.el_capitan.bottle.tar.gz
Warning: Bottle installation failed: building from source.
==> Downloading https://pypi.python.org/packages/source/p/platformio/platformio-2.8.4.tar.gz
######################################################################## 100.0%
==> Downloading https://pypi.python.org/packages/source/b/bottle/bottle-0.12.9.tar.gz
######################################################################## 100.0%
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec/vendor --single-version-externally-managed --record=installed.txt
==> Downloading https://pypi.python.org/packages/source/c/click/click-5.1.tar.gz
######################################################################## 100.0%
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec/vendor --single-version-externally-managed --record=installed.txt
==> Downloading https://pypi.python.org/packages/source/c/colorama/colorama-0.3.6.tar.gz
######################################################################## 100.0%
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec/vendor --single-version-externally-managed --record=installed.txt
==> Downloading https://pypi.python.org/packages/source/l/lockfile/lockfile-0.12.2.tar.gz
######################################################################## 100.0%
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec/vendor --single-version-externally-managed --record=installed.txt
Last 15 lines from /Users/spetrow/Library/Logs/Homebrew/platformio/04.python:
  File "/usr/local/lib/python2.7/site-packages/setuptools/command/build_py.py", line 59, in __getattr__
    self.data_files = files = self._get_data_files()
  File "/usr/local/lib/python2.7/site-packages/setuptools/command/build_py.py", line 72, in _get_data_files
    self.analyze_manifest()
  File "/usr/local/lib/python2.7/site-packages/setuptools/command/build_py.py", line 123, in analyze_manifest
    self.run_command('egg_info')
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/cmd.py", line 326, in run_command
    self.distribution.run_command(command)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py", line 972, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python2.7/site-packages/setuptools/command/egg_info.py", line 172, in run
    writer = ep.resolve()
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2386, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
ImportError: No module named pbr_json

READ THIS: https://git.io/brew-troubleshooting
```

```
00:43 $ pip install --upgrade pbr
Collecting pbr
  Downloading pbr-1.8.1-py2.py3-none-any.whl (89kB)
    100% |████████████████████████████████| 92kB 11.4MB/s
Installing collected packages: pbr
  Found existing installation: pbr 0.10.0
    Uninstalling pbr-0.10.0:
      Successfully uninstalled pbr-0.10.0
Successfully installed pbr-1.8.1
```

```
00:44 $ brew install platformio
==> Downloading https://homebrew.bintray.com/bottles/platformio-2.8.4.el_capitan.bottle.tar.gz

curl: (51) SSL: certificate verification failed (result: 5)
Error: Failed to download resource "platformio"
Download failed: https://homebrew.bintray.com/bottles/platformio-2.8.4.el_capitan.bottle.tar.gz
Warning: Bottle installation failed: building from source.
==> Downloading https://pypi.python.org/packages/source/p/platformio/platformio-2.8.4.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio-2.8.4.tar.gz
==> Downloading https://pypi.python.org/packages/source/b/bottle/bottle-0.12.9.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--bottle-0.12.9.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec/vendor --single-version-externally-managed --record=installed.txt
==> Downloading https://pypi.python.org/packages/source/c/click/click-5.1.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--click-5.1.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec/vendor --single-version-externally-managed --record=installed.txt
==> Downloading https://pypi.python.org/packages/source/c/colorama/colorama-0.3.6.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--colorama-0.3.6.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec/vendor --single-version-externally-managed --record=installed.txt
==> Downloading https://pypi.python.org/packages/source/l/lockfile/lockfile-0.12.2.tar.gz
Already downloaded: /Library/Caches/Homebrew/platformio--lockfile-0.12.2.tar.gz
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec/vendor --single-version-externally-managed --record=installed.txt
[... cut ...]
######################################################################## 100.0%
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec/vendor --single-version-externally-managed --record=installed.txt
==> python -c import setuptools... --no-user-cfg install --prefix=/usr/local/Cellar/platformio/2.8.4/libexec --single-version-externally-managed --record=installed.txt
🍺  /usr/local/Cellar/platformio/2.8.4: 503 files, 4.5M, built in 4 seconds
```

ivankravets, 2016-03-07:
@ip2k Hm. I don't have `pbr` in my system and `brew install platformio` works for me :(

I've pushed today new version 2.8.5. Do you have problems with upgrading?

ivankravets, 2016-03-08:
@ip2k 

Could you help us to resolve this issue with SSL and test these downloads?
```shell
cd /tmp
curl -L https://bintray.com/artifact/download/platformio/dl-packages/ldscripts-19.tar.gz -o ldscripts-19.tar.gz
curl -L https://www.geotrust.com/assets/images/SSL_Enterprise_Image.jpg -o SSL_Enterprise_Image.jpg
curl -L https://github.com -o gh.html
```

Thanks in advance!