platformio/platformio-core #396: "system param error" on latest platformio when flashing 8266

celsworth, 2015-12-21:
I noticed this on platformio 2.6.1 and 2.6.2 does the same thing:

platformio.ini:

```
[env:esp12]
platform = espressif
framework = arduino
board = esp12e
upload_speed = 230400
```

src/empty_sketch.cpp:

```
#include <ESP8266WiFi.h>

#include <WiFiClient.h>
#include <ESP8266mDNS.h>
#include <ArduinoOTA.h>

void setup()
{
    Serial.begin(115200);

    Serial.println("booting..");

    WiFi.persistent(false);
    WiFi.begin("xx", "xx");
    WiFi.mode(WIFI_STA);

    ArduinoOTA.begin();
}

unsigned long foo = 0;
void loop()
{
    ArduinoOTA.handle();

    if (millis() - foo > 10000)
    {
        Serial.println("I'm still alive :)\n");

        foo = millis();
    }
}
```

And I just get a reboot loop:

```
booting..

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont
sp: 3ffefd10 end: 3fff0100 offset: 01a0

>>>stack>>>
3ffefeb0:  3fff13b8 00000001 00000001 ffffff80
3ffefec0:  40216d6e 402247c8 00000018 3fff0db0
3ffefed0:  00000001 3fff1350 3fff1350 40216f58
3ffefee0:  3fff1350 3fff0db0 3ffe8481 00000000
3ffefef0:  3ffe8f80 3ffe8d94 fffc009a 3ffe8481
3ffeff00:  4022f858 00000064 f334fe1a fffeffff
3ffeff10:  00000018 0000ffff 00000000 00000001
3ffeff20:  3ffeff38 402134d4 3ffe8d10 3fff0db0
3ffeff30:  f334fe1a 3fff3634 00000000 00000000
3ffeff40:  00000000 00000000 00000000 00000000
3ffeff50:  00000000 00000000 00000000 00000000
3ffeff60:  00000000 00000000 402141df 3fff0db0
3ffeff70:  3fff01f4 3fff0db0 3fff0db0 40214227
3ffeff80:  00000000 3ffe8d94 3ffe8b2c 00000000
3ffeff90:  40206c91 00000003 00000003 3ffe8481
3ffeffa0:  40206dad 00000003 00000003 00000000
3ffeffb0:  00000000 40206e1a 00000003 3ffe8481
3ffeffc0:  40202a2a 00000000 00000000 00000000
3ffeffd0:  00000000 3ffe8489 3fff01f4 40202afd
3ffeffe0:  00000000 00000000 00000000 00000000
3ffefff0:  00000000 00000000 00000000 00000000
3fff0000:  00000000 00000000 00000000 00000000
3fff0010:  00000000 00000000 00000000 00000000
3fff0020:  00000000 00000000 00000000 00000000
3fff0030:  00000000 00000000 00000000 00000000
3fff0040:  00000000 00000000 00000000 00000000
3fff0050:  00000000 00000000 00000000 00000000
3fff0060:  00000000 00000000 00000000 00000000
3fff0070:  3ffef098 00000001 3ffe85e5 40223184
3fff0080:  3ffef098 00000008 3ffe847f 3ffef0cc
3fff0090:  3fffdc20 00000009 3ffef098 4020173d
3fff00a0:  3ffe85e4 00000000 3ffef098 4020173d
3fff00b0:  00000000 3fff08c0 3ffef098 40223258
3fff00c0:  0001c200 00000000 3ffef098 3ffef0cc
3fff00d0:  3fffdc20 00000000 3fff01f4 40205310
3fff00e0:  00000000 00000000 3ffef0c5 402018ba
3fff00f0:  00000000 00000000 3ffef0e0 40202948
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16
tail 0
chksum 0x42
csum 0x42
~ld
   system param error
c9 ff 00 00 78 e5 fe 3f 61 6a f1 51 15 fa c0 20 00 42 25 96 60 44 20 c0 20 00 31
 fa ff 42 65 96
22 63 7f 0d f0 00 00 00 12 c1 f0 29 11 09 01 01 4d ed c0 00 00 08 11 61 0a fa c0 20 00 22 26 93
c0 20 00 52 26 92 c0 20 00 72 26 93 77 12 07 2d 07 c0 20 00 52 26 92 5a 00 07 35 01 1b 22 c0 20
00 02 66 b3 c0 20 00 22 66 b4 21 51 f1 c0 20 00 02 26 b2 20 00 20 c0 20 00 02 66 b2 01 23 ee c0
00 00 08 01 12 c1 10 0d f0 00 00 00 70 e7 fe 3f 12 c1 f0 09 01 01 fd ff 00 02 a0 28 00 8c 12 c0
[snip, there is a lot more of this, let me know if it woudl be useful]

<repeat>
```

The same code works on 2.5.0 and earlier, its just an empty sketch that doesn't do much after all. Any ideas? Has the ldscript changed significantly? I can give you a compiled binary if it will help.

ivankravets, 2015-12-21:
Did you update `ldscripts`? They were changed significantly.

celsworth, 2015-12-21:
```
mint-vmware:~/src/arduino/empty_sketch> platformio upgrade
You're up-to-date!
PlatformIO 2.6.2 is currently the newest version available.
mint-vmware:~/src/arduino/empty_sketch> platformio update

Platform atmelavr
--------
Updating toolchain-atmelavr package:
Versions: Current=4, Latest=4 	 [Up-to-date]
Updating tool-avrdude package:
Versions: Current=3, Latest=3 	 [Up-to-date]
Updating framework-arduinoavr package:
Versions: Current=23, Latest=23 	 [Up-to-date]
Updating tool-micronucleus package:
Versions: Current=1, Latest=1 	 [Up-to-date]

Platform espressif
--------
Updating ldscripts package:
Versions: Current=18, Latest=18 	 [Up-to-date]
Updating sdk-esp8266 package:
Versions: Current=5, Latest=5 	 [Up-to-date]
Updating tool-esptool package:
Versions: Current=6, Latest=6 	 [Up-to-date]
Updating framework-arduinoespressif package:
Versions: Current=7, Latest=7 	 [Up-to-date]
Updating toolchain-xtensa package:
Versions: Current=2, Latest=2 	 [Up-to-date]
Updating [ 1 ] OneWire library:
Versions: Current=ef5e5c64cb, Latest=ef5e5c64cb 	 [Up-to-date]
Updating [ 54 ] DallasTemperature library:
Versions: Current=6087c1b775, Latest=6087c1b775 	 [Up-to-date]
```

all seems up to date?

ivankravets, 2015-12-21:
Yes, now please upgrade to 2.6.2. Thanks.

celsworth, 2015-12-21:
This is 2.6.2, there is nothing to update?

commit 7b1b9087 (2015-12-21): Fix invalid ROM size for ESP8266-based boards // Resolve #396
closed 2015-12-21 by ivankravets