platformio/platformio-core #397: simulAVR and gdb-avr missed in toolchain-atmelavr-windows

CraigJPerry, 2015-12-21:
Like #334, simulavr & gdb is not installed on Windows 10, x64.

Tried updating:

    [windows10x64]$ platformio update
    
    Platform atmelavr
    --------
    Updating toolchain-atmelavr package:
    Versions: Current=2, Latest=2    [Up-to-date]
    Updating tool-avrdude package:
    Versions: Current=2, Latest=2    [Up-to-date]
    Updating framework-arduinoavr package:
    Versions: Current=23, Latest=23          [Up-to-date]

Tried dev version of platformio and updating toolchains:

    [windows10x64]$ platformio --version
    PlatformIO, version 2.6.2.dev0

valeros, 2015-12-21:
Hi @CraigJPerry 
`avr-gdb` was already in `.platformio\toolchain-atmelavr\bin`
Anyway, I've just added `simul-avr to package`, please update via `platformio update`

CraigJPerry, 2015-12-23:
Awesome, thanks @valeros i confirm it works for me now:

    $ platformio update
    There is a new version 2.6.3 of PlatformIO available
    ...

    Updating toolchain-atmelavr package:
    Versions: Current=2, Latest=3    [Out-of-date]
    Uninstalling toolchain-atmelavr package:        [OK]
    Installing toolchain-atmelavr package:
    Downloading
    Unpacking
    ...

    $ ls -l .platformio/packages/toolchain-atmelavr/bin/simulavr.exe
    -rwxr-xr-x 1 craig 197609 2359296 Jan 19  2010 .platformio/packages/toolchain-at                          melavr/bin/simulavr.exe*

ivankravets, 2015-12-23:
@CraigJPerry thanks for the report. If you like PlatformIO we would be thankful for the :star: https://github.com/platformio/platformio/stargazers

closed 2015-12-23 by ivankravets

rahul30694, 2017-08-23:
I m using Ubuntu 16.04 and have updated platformio via platformio update.But also i cant find avr-simul.
Please help me out