platformio/platformio-core #398: Support for Sming ESP8266 framework

ivankravets, 2015-12-22:
@SmingHub is open Source framework for high efficiency native ESP8266 development

https://github.com/SmingHub/Sming

itsjustvenky, 2015-12-29:
This will be a great addition to PlatformIO

ivankravets, 2016-01-01:
@eerimoq and @kireevco, could you help us to add support for Sming framework? 

Thanks in advance!

chall3ng3r, 2016-01-20:
Yes, I've been developing using Sming framework for ESP8266. It's awesome. Its also showing up in monthly top downloads on PlatformIO stats.

kireevco, 2016-01-20:
@ivankravets so, what do we need from sming side? make the projects work under scons, CMAKE? Or something else?

itsjustvenky, 2016-01-20:
@kireevco When I build my esp-open-sdk, it is taking 3.5GB of disk space and Sming depends on this SDK. So here is my question, do we really need 3.5GB files for Sming or I am doing something wrong after build not cleanin-up. ? thank you.

kireevco, 2016-01-20:
@itsjustvenky you can try binary builds of [esp-alt-sdk](https://github.com/kireevco/esp-alt-sdk)

itsjustvenky, 2016-01-20:
@kireevco  Thank you. Sure I will give a try.

voyo, 2016-03-07:
I just discovered PlatformIO. Looks impressive, will try to dive deeper on this.
In meantime +1 for Sming framework support here. Is is possible ? when can we expect this ?

ivankravets, 2016-03-07:
@voyo we wait when someone will help with it. Need to convert "make-flow" to PlatformIO Build System. Here are examples: https://github.com/platformio/platformio/tree/develop/platformio/builder/scripts/frameworks

robotiko, 2016-03-07:
@ivankravets,
it will be great to take advantage of PlatformIO in Sming.  It is just impressive. 
I had a fast look at the framework scripts to get an idea of what will it require to add Sming as framework. 
Any general guideline? I checked the doc and the scripts (not familiar at all with python, but that should be a minor detail) and don't get a clear pic of what is needed.  In this special case, the platform and boards would be the same previously defined for Arduino and Espressif, so the toolchain is already there.  Several frameworks for the same platform and boards would be an issue?   What would be the best reference script to start? 
What is what PlatformIO needs to be defined in the builder?

Thxs

ivankravets, 2016-03-08:
@robotiko please take a look at my explanation for @hreintke https://gitter.im/platformio/platformio?at=56ad236f586242210ae08534

ivankravets, 2016-11-27:
Moved to https://github.com/platformio/platform-espressif8266/issues/17

closed 2016-11-27 by ivankravets