platformio/platformio-core #399: Incorrect readings from DS18B20 sensor on espressif

jasonmhite, 2015-12-22:
I've got a DS18B20 temperature sensor hooked up to my ESP8266, which I'm accessing via the DallasTemperature library (number 54). I get junk readings of the sensor *only* when building with platformio. There are no error messages during the build and it flashes fine and I can do things like printing to the serial console, so I know it's not crashing or something similar... it's just the sensor readings are junk. However, if I build the exact same code using the Arduino IDE it works no-problem and prints a correct temperature reading.

## Example code
```
#include <OneWire.h>
#include <DallasTemperature.h>

#define BAUD 115200
#define DS18_PIN 5

OneWire oneWire(DS18_PIN);
DallasTemperature sensor(&oneWire);

void setup() {
    Serial.begin(BAUD);

    sensor.begin();
}

void loop() {
    sensor.requestTemperatures();
    Serial.println(sensor.getTempCByIndex(0));
    delay(1000);
}
```
Specifically, when built with platformio, this will print `-127.00` over and over, wheras with the Arduino IDE it gives the proper temperature reading. Note that this is the same value it prints if you give it the wrong pin number (as in, a pin that isn't connected to anything).

I've also experienced essentially the exact same issue with a DHT22 temp/humidity sensor, where the same code would work with Arduino and not with platformio, but I don't have a code snippet handy. I would *love* to get this working as I hate the Arduino IDE with a passion, but this issue is killing me.

## Versions

* platformio - 2.6.3. It wasn't working with the version of the espressif tools I pulled down about two weeks ago and I also just updated today to the latest versions with `platformio platforms update`. (Ref: today is 2015-12-22)
* Arduino IDE - 1.6.5
* Boards definition for ESP8266 for Arduino IDE - 2.0.0-rc2
* The actual module - I have had this issue with both an ESP-07 and an ESP-12E.

Thanks, lemme know if you need more details.

ivankravets, 2015-12-22:
Do you mean that use the latest PlatformIO and packages? Could you try to 

* `pip install -U platformio`
* `rm -r ~/.platformio`

And then build your project. Thanks.

jasonmhite, 2015-12-22:
That's what I meant, I was using the latest version.
Just to be sure I deleted `~/.platformio` and redownloaded the espressif platform, then cleaned and rebuilt my project. It didn't fix it.

jasonmhite, 2015-12-22:
Has anybody else gotten a DS18 or DHT22 working on the ESP8266 with platformio? They're both pretty common sensors and since I have issues with both the ESP-07 and ESP-12E it makes me think that it's not only me and there's some sort of software glitch, but I don't mind describing the specifics of my setup if you think it's something specific about the way I'm using the sensors.

ivankravets, 2015-12-22:
@jasonmhite please wait for @valeros. He will compare both firmwares (Arduino IDE vs PlatformIO). Could you provide him simple sketch which works with Arduino IDE but doesn't work with PlatformIO?

jasonmhite, 2015-12-22:
@ivankravets No problem, just speculating about the cause.

The code snippet I put in my first post is a minimal example. With the Arduino IDE it prints the correct temperature values, but compiling the same code with Platformio it prints `-127.00` (the same value it prints as when you initialize the DS18 library on an empty/wrong pin).

valeros, 2015-12-24:
Hi @jasonmhite 
Does your sketch work fine with stable framework 2.0.0 (not 2.0.0-rc2)?

jasonmhite, 2015-12-29:
Just wanted to say I haven't forgotten about this, but I'm on travel until Jan 1. I'll try the release version of the 2.0 framework then.

jasonmhite, 2016-01-03:
Alright, I got back to my workbench and tested it using the 2.0.0 final release of the ESP8266 board definition in the Arduino IDE. It works fine.

jasonmhite, 2016-01-26:
So an update on this... the problem has disappeared and now things are working with some new ESPs I got. I *think* I might have gotten a bad batch of ESPs or DS18s; I tried with several different ESPs and DS18s and they all did not work, but all the ones I tried were from the same orders. I got some new ones and they work, so I guess I got a crop of faulty parts. That doesn't really explain why they worked with the Arduino IDE and not PIO, but I guess it was just an unlucky combination of something going wrong. In any case, it works now and I am going to close the issue. Thanks for looking at this.

closed 2016-01-26 by jasonmhite

ivankravets, 2016-01-26:
Thanks for the report! Please give us a star :star: https://github.com/platformio/platformio/stargazers

Lassska, 2020-02-19:
https://github.com/esp8266/Arduino/issues/4940#issuecomment-588150434

Lassska, 2020-02-20:
After 10 reads - very stable median values for DS18B20 with ESP8266 Web Server
Data wire is connected to GPIO 2
Show one decimal in the web server (see attach)
EXAMPLE: // tempC+/-x.xx calibration value =+0.50 celsius
// tempF+/-x.xx calibration value =+0,82 fahrenheit

04:14:49.453 ->
04:14:49.487 -> Connecting to WiFi
04:14:49.977 -> ..
04:14:50.485 -> 192.168.0.106
04:14:51.241 -> Temperature Celsius: 20.79
04:14:51.309 -> Temperature Fahrenheit: 69.46
04:14:51.377 -> Temperature Celsius: 20.78
04:14:51.445 -> Temperature Fahrenheit: 69.45
04:14:51.648 -> Temperature Celsius: 20.77
04:14:51.716 -> Temperature Fahrenheit: 69.44
04:15:02.128 -> Temperature Celsius: 20.77
04:15:02.162 -> Temperature Fahrenheit: 69.43
04:15:12.127 -> Temperature Celsius: 20.76
04:15:12.161 -> Temperature Fahrenheit: 69.42
04:15:14.099 -> Temperature Celsius: 20.76
04:15:14.133 -> Temperature Fahrenheit: 69.41
04:15:19.505 -> Temperature Celsius: 20.74
04:15:19.539 -> Temperature Fahrenheit: 69.41
04:15:24.972 -> Temperature Celsius: 20.74
04:15:25.006 -> Temperature Fahrenheit: 69.40
04:15:35.167 -> Temperature Celsius: 20.73
04:15:35.201 -> Temperature Fahrenheit: 69.40
04:15:41.778 -> Temperature Celsius: 20.73
04:15:41.812 -> Temperature Fahrenheit: 69.36
04:15:52.029 -> Temperature Celsius: 20.74
04:15:52.064 -> Temperature Fahrenheit: 69.36
04:15:54.678 -> Temperature Celsius: 20.74
04:15:54.745 -> Temperature Fahrenheit: 69.36
04:16:04.870 -> Temperature Celsius: 20.73
04:16:04.904 -> Temperature Fahrenheit: 69.37
04:16:14.859 -> Temperature Celsius: 20.73
04:16:14.927 -> Temperature Fahrenheit: 69.35
04:16:24.869 -> Temperature Celsius: 20.73

CODE:
`/******************/

// Import required libraries
#ifdef ESP32
#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#else
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <Hash.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#endif
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is connected to GPIO 2
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature sensor
DallasTemperature sensors(&oneWire);

// Replace with your network credentials
const char* ssid = "TP-Link_852E";
const char* password = "15866493";

[... 103 lines cut ...]

// Print ESP Local IP Address
Serial.println(WiFi.localIP());

// Route for root / web page
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
request->send_P(200, "text/html", index_html, processor);
});
server.on("/temperaturec", HTTP_GET, [](AsyncWebServerRequest *request){
request->send_P(200, "text/plain", readDSTemperatureC().c_str());
});
server.on("/temperaturef", HTTP_GET, [](AsyncWebServerRequest *request){
request->send_P(200, "text/plain", readDSTemperatureF().c_str());
});
// Start server
server.begin();

//populeaza vectorul de medie
for(int i=0;i<LEN_VECT;i++) vect[i] = sensors.getTempCByIndex(0);
for(int i=0;i<LEN_VECT2;i++) vect2[i] = sensors.getTempFByIndex(0);
}

void loop(){

}`