platformio/platformio-core #400: Digispark-tiny upload error

fitzterra, 2015-12-23:
Trying to upload to a digispark-tiny (arduino, attiny85) via micronuclues fails with:

```
$ platformio run -e digispark-tiny -t upload
[Wed Dec 23 08:27:59 2015] Processing digispark-tiny (platform: atmelavr, board: digispark-tiny, framework: arduino)
-----------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/digispark-tiny/firmware.hex"])
Please unplug/plug device ...
"/home/tomc/.platformio/packages/tool-micronucleus/micronucleus" -c digispark --timeout 60 .pioenvs/digispark-tiny/firmware.hex
sh: 1: /home/tomc/.platformio/packages/tool-micronucleus/micronucleus: not found
scons: *** [upload] Error 127
```
There is no pause at the point where the message `Please unplug/plug device ...` comes up and the error appears immediately.

This is on a Xubuntu 15.10 64-bit machine with PlatformIO installed inside a `virtualenv`.

Looking at the micronucleus binary, it seems it is a 32-bit version. Not sure if this is the issue:

```
$ file /home/tomc/.platformio/packages/tool-micronucleus/micronucleus
/home/tomc/.platformio/packages/tool-micronucleus/micronucleus: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=37731088830509f6a90d9765e9784239bb32ffc8, not stripped
```

Anyone seen this before or have any advice?

Thanks,
 Tom

ivankravets, 2015-12-23:
@valeros please check that this 32-bit binary works on 64-bit host OS

ivankravets, 2015-12-23:
Similar issue #394

valeros, 2015-12-23:
Hi @fitzterra 
Could you test upload with this micronucleus:
https://dl.dropboxusercontent.com/u/10025728/tool-micronucleus.zip

fitzterra, 2015-12-24:
I got it to work by doing the following:
* installing `lib32stdc++6` from the package manager - I doubt if this has anything to do with the result, but saw somewhere that it may be needed for the USB support
* copying the 64-bit micronucleus from the Digistump v2.04a library installed by the Arduino library installer.

Here are my current `~/.platformio/packages/tool-micronucleus` dir:

```
$ ls
micronucleus  micronucleus.orig
```

The `.orig` version is the original installed by `platformio` and the other is the one from Digistump:

```
$ file *
micronucleus:      ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=cdbba5533de04828c8a94ab502d6effab96a7a4c, not stripped
micronucleus.orig: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=37731088830509f6a90d9765e9784239bb32ffc8, not stripped
```

Does this help?

valeros, 2015-12-24:
@fitzterra Please, update your `tool-micronucleus` package via `platformio update`

fitzterra, 2015-12-24:
@valeros A great big thank you! After the update I now have a 64-bit micronucleus and the `unplug/replug` pause and timeout seems to work perfectly. I can only test and actual flash tomorrow and will report here if it does not work, but am confident the issue is now resolved.

A great big thank you for your effort on Christmas Eve (or Christmas Day, depending where you are :-) ) - much appreciated.

ivankravets, 2015-12-24:
@fitzterra please reopen this issue if it doesn't work.

ivankravets, 2015-12-24:
@fitzterra I would recommend you to gift @valeros :star: here https://github.com/platformio/platformio/stargazers. He likes them :blush: :christmas_tree:

closed 2015-12-24 by ivankravets