platformio/platformio-core #401: Use current head of ESP8266 Arduino

sticilface, 2015-12-23:
Is there a way to use the current head from 
 https://github.com/esp8266/Arduino 

see here for my issues trying to do this (Posted in the wrong thread by mistake) 
https://github.com/platformio/platformio/issues/366#issuecomment-166909856

Dropping in the latest code compiles but does not link correctly.

ivankravets, 2015-12-23:
Firstly, you need to gift :star: for @valeros here https://github.com/platformio/platformio/stargazers :smile:

ivankravets, 2015-12-23:
I want to note that we don't support ALPHA/BETA/STAGE or other development software because could spent spare time only the ONE framework or platform. However, I'll give a few hints which should allow you to use latest ESP8266 for Arduino.

1. Clone repo to `~/.platformio/packages/framework-arduinoespressif` and keep `version.txt` file (you can use it from stable version)
2. Try to build project
3. If you have build errors, then try your project using the same `stage` on Arduino IDE. 
4. If it works with Arduino IDE but doesn't work with PlatformIO, then we wait from you:
    - test project/files
    - detailed log of build process from Arduino IDE (please copy it from console to http://pastebin.com)
    - detailed log of build process from PlatformIO (please copy it from console to http://pastebin.com)

sticilface, 2015-12-23:
Many thanks :)

I've done all these.. actions.. I will provide what you need.... 

Arduino Output http://pastebin.com/6cKrVnKS
PlatformIO Output http://pastebin.com/Tr1cQv8F

It fails at the linking stage, right at the end!  

compiling any sketch produces it.. The one I use us 
http://pastebin.com/QtfTWEsT

Your help is much appreciated... and I know others will be helped too.  The IDE moves at such a pace at the moment...  but i do understand that you can't support it!

sticilface, 2015-12-23:
I should add that this is option 4.  compile with Arduino IDE, but not with PlatformIO.

ivankravets, 2015-12-23:
Try to find difference between 
```
Arduino IDE
-lm -lgcc -lhal -lphy -lpp -lnet80211 -llwip -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh

PlatformIO
-lsmartconfig -lpp -lmain -lwpa -llwip -lnet80211 -lwps -lcrypto -lphy -lhal -laxtls -lgcc -lm 
```

:)

[build_flags](http://docs.platformio.org/en/latest/projectconf.html#build-flags) should help you.

sticilface, 2015-12-23:
Even with my PhD I spotted that one. So far though my attempts at build_flags has failed:( 

I've added -lmesh to lots of the options but it still fails. I'm a beginner! Never used anything apart from arduino IDE... Another clue please...

ivankravets, 2015-12-23:
Don't make any modifications to your projects. The problem is in LD script. I've just created separated branch with LD scripts. Please use it https://github.com/platformio/platformio-pkg-ldscripts/tree/esp8266_stage

ivankravets, 2015-12-23:
I updated comment https://github.com/platformio/platformio/issues/401#issuecomment-166958952

sticilface, 2015-12-24:
That worked purrrrfectly.  Thank you

donnib, 2016-01-15:
@ivankravets I am trying to build my project using latest code from Arduino/ESP8266 and by following your guide above i can build but the board keeps on rebooting, i tried Arduino IDE with the same code and it works there so i guess maybe there is a problem with the flashing ? I know you said you don't support these versions so if you have an idea that would be good to hear if not i will see what i can do.

ivankravets, 2016-01-15:
@donnib please follow to step 5 here https://github.com/platformio/platformio/issues/401#issuecomment-166958952

donnib, 2016-01-17:
@ivankravets ok so here it is
https://gist.github.com/donnib/8b5ecff157a7cf129daf

I have posted there code used for Arduino IDE and PlatformIO although i had make changes in the code to compile in Arduino which was weird. I have a method readFromRTC() which Arduino IDE complained was not found so i moved it to the top then it compiled. PlatformIO compiled fine without moving the method. I have not tried to build the code used in Arduino IDE where the method is moved in PlatformIO. I have also pasted the console output from PlatformIO and ArduinoIDE. Do you need the bin files ?

valeros, 2016-01-18:
Hi @donnib
I've just tested latest ESP8266 framework with my `esp01` board.
Our example `wifi-scan` works without any errors.
Serial output from your sketch:
```
Reason for reboot 0
I am in setup...
We just been booted first time so disable WiFi to lower power usage
rtc = 0
```

donnib, 2016-01-18:
@valeros thx well that's weird. I guess i must be doing something wrong then....hmmm. i wonder what that could be. Any ideas ?

valeros, 2016-01-18:
@donnib  Yes, latest master branch https://github.com/esp8266/Arduino/archive/master.zip
Could you try to upload simple [wifi-scan](https://github.com/platformio/platformio/tree/develop/examples/espressif/esp8266-wifiscan)  example?

donnib, 2016-01-18:
@valeros yes i just tried and that worked fine but my code not. I have even tried to run on a different workstation so now i tried windows and MacOSX. Can it be some HW related thing ? Idk what that could be when it works in Arduino IDE

valeros, 2016-01-19:
@donnib According to your Arduino IDE build log the firmware was compiled for 40MHz frequency. Could you try [latest develop version of PlatformIO](http://docs.platformio.org/en/latest/installation.html#d-development-version) and specify `board_f_cpu=40000000L` option in your `platformio.ini`?

donnib, 2016-01-20:
@valeros ok so i did what you suggested. I uninstalled platformio as per the doc, installed dev, added the ```board_f_cpu``` line and then i flashed and wola it worked, i was curios so i removed the board line from the platform.ini, it still worked, hmmm. I then uninstalled dev platformio and installed stable and it still worked, now i don't understand what happened, what made it work ?

Coaxial, 2016-02-20:
Hi,

This code doesn't build with platformio following your instructions to use Arduino-git but it builds using the Arduino framework's git version with the Arduino IDE.

Problematic code: https://github.com/Coaxial/vintage-cec/tree/tmp/ws-build-issue
Platformio output:
```
$ make
platformio -f -c vim run
[Sat Feb 20 15:04:39 2016] Processing nodemcu (platform: espressif, lib_install: 549, board: nodemcu, framework: arduino)
-----------------------------------------------------------------------------------------------------
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/src/main.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=210 -DPLATFORMIO=020804 -I/Users/hfp/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant -I.pioenvs/nodemcu/IRremoteESP8266 -I.pioenvs/nodemcu/WebSockets_ID549 src/main.cpp
xtensa-lx106-elf-ar rcs .pioenvs/nodemcu/libFrameworkArduino.a
xtensa-lx106-elf-ranlib .pioenvs/nodemcu/libFrameworkArduino.a
xtensa-lx106-elf-g++ -o .pioenvs/nodemcu/IRremoteESP8266/IRremoteESP8266.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DARDUINO=210 -I/Users/hfp/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/nodemcu/FrameworkArduino -I.pioenvs/nodemcu/FrameworkArduinoVariant -I.pioenvs/nodemcu/IRremoteESP8266 -I.pioenvs/nodemcu/WebSockets_ID549 .pioenvs/nodemcu/IRremoteESP8266/IRremoteESP8266.cpp
In file included from .pioenvs/nodemcu/FrameworkArduino/Arduino.h:38:0,
from .pioenvs/nodemcu/IRremoteESP8266/IRremoteInt.h:27,
from .pioenvs/nodemcu/IRremoteESP8266/IRremoteESP8266.cpp:29:
.pioenvs/nodemcu/FrameworkArduino/esp8266_peri.h:24:21: fatal error: c_types.h: No such file or directory
#include "c_types.h"
^
compilation terminated.
In file included from .pioenvs/nodemcu/FrameworkArduino/Arduino.h:38:0,
from src/main.cpp:1:
.pioenvs/nodemcu/FrameworkArduino/esp8266_peri.h:24:21: fatal error: c_types.h: No such file or directory
#include "c_types.h"
^
compilation terminated.
scons: *** [.pioenvs/nodemcu/IRremoteESP8266/IRremoteESP8266.o] Error 1
scons: *** [.pioenvs/nodemcu/src/main.o] Error 1
==================================== [ ERROR ] Took 1.19 seconds ====================================
make: *** [all] Error 1
```

Arduino IDE output:
```

Sketch uses 241,676 bytes (23%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 33,761 bytes (41%) of dynamic memory, leaving 48,159 bytes for local variables. Maximum is 81,920 bytes.
```

ivankravets, 2016-03-05:
Moved information to the official docs http://docs.platformio.org/en/latest/platforms/espressif.html#using-arduino-framework-with-staging-version

closed 2016-03-05 by ivankravets

majklovec, 2016-08-31:
After going through your steps. The first run is compiled ok. But the second run of `pio run` "updates" to the last stable version

Is there any way to deny the update?

ivankravets, 2016-09-14:
See my comment here https://community.platformio.org/t/platformio-3-0-broke-all-my-esp8266-projects/747/7?u=ivankravets

commit 213938a9 (2016-03-05): Add info about "Using Arduino Framework with Staging version" // Resolve #401