platformio/platformio-core #403: Add board type/support for Adafruit 32u4 based Feather boards

chrisbalmer, 2015-12-24:
I tried using the flora8 board settings for an Adafruit Feather BLE board as they have the same MCU (32u4, 8Mhz, 32K flash, 2K ram, 3.3v). However during the upload, the serial port disappears briefly and platformio reports that the port doesn't exist.

**Build Attempt**

```
$ platformio run --target upload
[Wed Dec 23 23:15:48 2015] Processing flora8 (platform: atmelavr, upload_port: /dev/cu.usbmodem1411, board: flora8, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/flora8/firmware.hex"])
"/Users/user/.platformio/packages/tool-avrdude/avrdude" -v -p atmega32u4 -C "/Users/user/.platformio/packages/tool-avrdude/avrdude.conf" -c avr109 -b 57600 -P /dev/cu.usbmodem1411 -D -U flash:w:.pioenvs/flora8/firmware.hex:i

avrdude: Version 6.0.1, compiled on Dec 16 2013 at 17:26:24
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/Users/user/.platformio/packages/tool-avrdude/avrdude.conf"
EUser configuration file is "/Users/user/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/cu.usbmodem1411
Using Programmer              : avr109
Overriding Baud Rate          : 57600
avrdude: ser_open(): can't open device "/dev/cu.usbmodem1411": No such file or directory
avrdude: butterfly_recv(): programmer is not responding
scons: *** [upload] Error 1
====================================================================================== [ ERROR ] Took 6.91 seconds ======================================================================================
$
```

**During Build Serial Port Disappears**

```
$ ls /dev/cu*
/dev/cu.Bluetooth-Incoming-Port	/dev/cu.usbmodem1411
$ ls /dev/cu*
/dev/cu.Bluetooth-Incoming-Port
$ ls /dev/cu*
/dev/cu.Bluetooth-Incoming-Port	/dev/cu.usbmodem1411
$
```

If I init the project as a "sparkfun_promicro8" board (same MCU according to the boards list), it will upload successfully.

OS: OS X El Capitan

ivankravets, 2015-12-24:
Please install http://docs.platformio.org/en/latest/installation.html#d-development-version and test it. Don't forget to `platformio update`.

closed 2015-12-24 by ivankravets

chrisbalmer, 2015-12-25:
Works perfectly now! Thanks!

commit dfe39f38 (2015-12-24): Add support for the new Adafruit boards Bluefruit Micro and Feather // Resolve #403