platformio/platformio-core #405: unable to install platformio

grzegorzshub, 2015-12-25:
Hello
I'm trying to install the platormio and keep getting the same error:
sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"

==> Installing Python Package Manager ...
Requirement already up-to-date: pip in /Library/Python/2.7/site-packages
[SUCCESS]

==> Installing PlatformIO and dependencies ...
Collecting platformio
  Downloading platformio-2.6.3.tar.gz (85kB)
Requirement already up-to-date: bottle in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: click<6,>=3.2 in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: lockfile>=0.9.1 in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: pyserial in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: requests>=2.4.0 in /Library/Python/2.7/site-packages (from platformio)
Collecting scons (from platformio)
  Downloading scons-2.4.1.tar.gz (639kB)
Installing collected packages: scons, platformio
  Found existing installation: scons 2.3.6
    Uninstalling scons-2.3.6:
      Successfully uninstalled scons-2.3.6
  Running setup.py install for scons
    Complete output from command /usr/bin/python -c "import setuptools, tokenize;__file__='/private/tmp/pip-build-JIjG2m/scons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-GRoiPe-record/install-record.txt --single-version-externally-managed --compile:
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help
    
    error: option --single-version-externally-managed not recognized
    
    ----------------------------------------
  Rolling back uninstall of scons

Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/private/tmp/pip-build-JIjG2m/scons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-GRoiPe-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/tmp/pip-build-JIjG2m/scons

[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio/issues >

ivankravets, 2015-12-25:
Hi,

Try this
```
sudo pip install -U scons
sudo pip install -U platformio
```

grzegorzshub, 2015-12-25:
Hi
I'm sorry for hassle, just found:
http://docs.platformio.org/en/latest/faq.html#pip-scons-error-option-single-version-externally-managed-not-recognized
Reinstalling scons sorted the issue out.
Thank you, much appreciate.
Best

ivankravets, 2015-12-25:
Did you resolve it?

grzegorzshub, 2015-12-25:
yes, all is working fine so far

ivankravets, 2015-12-25:
Could you give us a :star: here https://github.com/platformio/platformio/stargazers? Thanks!

ivankravets, 2015-12-25:
Thanks! Happy Holidays and Merry Christmas! :christmas_tree:

commit 27f32667 (2021-09-16): Fixed a "KeyError: Invalid board option 'build.cpu'" when using a precompiled library with a board that does not have a CPU field in the manifest // Resolve #405
closed 2015-12-25 by ivankravets