platformio/platformio-core #408: Unit testing for embedded

ivankravets, 2015-12-28:
The issue is addition to #258. Requested by @Oaz, @kireevco.

How it will work? Nothing new here, general [Unit testing design](https://en.wikipedia.org/wiki/Unit_testing)

1. Developer writes tests (using assert function/macro TBD) and place them into `tests` folder
2. Run `platformio test` command

PlatformIO will use each `test` as source code for the new project and the source code of existing project will be used as input library. Developer writes tests for the specific part of code, PlatformIO uploads it on-the-fly, tests, gathers results, shows reports to the developer.

Comments are welcome.

ivankravets, 2015-12-28:
Tip: @ThrowTheSwitch https://github.com/ThrowTheSwitch/Unity

eerimoq, 2015-12-30:
This is how I normally writes tests: https://github.com/eerimoq/simba/blob/master/tst/kernel/event/main.c. It is very basic and works very well for me. There are plenty of examples in the tst-folder in that repository.

ivankravets, 2015-12-30:
@eerimoq could you provide test-output from AVR?

eerimoq, 2015-12-30:
Tonight, when I get home. I'll add output for Arduino Nano and Arduino Due. It's pretty much the same as the above though. Ideally, the output should be the same independently of what hardware it is executed on. Except cases where a device has too little memory and the whole test suite will not fit.

ivankravets, 2015-12-30:
@eerimoq I agree with you. Now need to decide how to implement it: write own UNIT Tests framework or use existing solutions, like @ThrowTheSwitch https://github.com/ThrowTheSwitch/Unity

eerimoq, 2015-12-30:
It's probably useful for many programmers to get the test framework generated when the create  a project, but, also, I think platformio should provide a high level API that makes it possible to communicate with platformio with any test framework.

ivankravets, 2015-12-30:
Good. I'll wait output from AVR platform

eerimoq, 2015-12-30:
**Arduino Nano**

    erik@erik-PC ~/simba/simba/tst/kernel/event
    $ make -s BOARD=arduino_nano test
    Generating settings.bin from ../../../make/settings.ini
    Generating settings.h from ../../../make/settings.ini
    Compiling main.c
    Compiling ../../../src/boards/arduino_nano/board.c
    Compiling ../../../src/mcus/atmega328p/mcu.c
    Compiling ../../../src/kernel/chan.c
    Compiling ../../../src/kernel/event.c
    Compiling ../../../src/kernel/fs.c
    Compiling ../../../src/kernel/log.c
    Compiling ../../../src/kernel/queue.c
    Compiling ../../../src/kernel/sem.c
    Compiling ../../../src/kernel/setting.c
    Compiling ../../../src/kernel/shell.c
    Compiling ../../../src/kernel/std.c
    Compiling ../../../src/kernel/sys.c
    Compiling ../../../src/kernel/thrd.c
    Compiling ../../../src/kernel/time.c
    Compiling ../../../src/kernel/timer.c
    Compiling ../../../src/drivers/adc.c
    Compiling ../../../src/drivers/ds18b20.c
    Compiling ../../../src/drivers/ds3231.c
    Compiling ../../../src/drivers/exti.c
    Compiling ../../../src/drivers/mcp2515.c
    Compiling ../../../src/drivers/nrf24l01.c
    Compiling ../../../src/drivers/spi.c
    Compiling ../../../src/drivers/owi.c
    Compiling ../../../src/drivers/pin.c
    Compiling ../../../src/drivers/pwm.c
    Compiling ../../../src/drivers/sd.c
    Compiling ../../../src/drivers/uart.c
    Compiling ../../../src/drivers/uart_soft.c
    Compiling ../../../src/slib/crc.c
    Compiling ../../../src/slib/fat16.c
    Compiling ../../../src/slib/harness.c
    Compiling ../../../src/slib/hash_map.c
    Compiling ../../../src/slib/midi.c
    Compiling gen/simba_gen.c
    Linking event_suite.out
    Running event_suite.out
    avrdude -p atmega328p -D -P COM3 -c arduino -V -b 57600 -U eeprom:w:settings.bin:r
    
    avrdude.exe: AVR device initialized and ready to accept instructions
    
    Reading | ################################################## | 100% 0.00s
    
    avrdude.exe: Device signature = 0x1e950f
    avrdude.exe: reading input file "settings.bin"
    avrdude.exe: writing eeprom (1024 bytes):
    
    Writing | ################################################## | 100% 8.19s
    
    avrdude.exe: 1024 bytes of eeprom written
    
    avrdude.exe done.  Thank you.
    
    avrdude -p atmega328p -D -P COM3 -c arduino -V -b 57600 -U flash:w:event_suite.hex
    
    avrdude.exe: AVR device initialized and ready to accept instructions
    
    Reading | ################################################## | 100% 0.00s
    
    avrdude.exe: Device signature = 0x1e950f
    avrdude.exe: reading input file "event_suite.hex"
    avrdude.exe: input file event_suite.hex auto detected as Intel Hex
    avrdude.exe: writing flash (22476 bytes):
    
[... 77 lines cut ...]
    Erase flash
    Write 40888 bytes to flash
    [==============================] 100% (160/160 pages)
    Set boot flash true
    app:   event_suite-0.0.0 built 2015-12-30 17:55 CET by erik.
    board: Arduino Due
    mcu:   Atmel SAM3X8E Cortex-M3 @ 84MHz, 96k sram, 512k flash
    enter: test_read_write
    exit: test_read_write: PASSED
    enter: test_poll
    exit: test_poll: PASSED
    enter: test_poll_timeout
    exit: test_poll_timeout: PASSED
                NAME           PARENT        STATE  PRIO   CPU  MAX-STACK-USAGE  LOGMASK
                main                       current     0    0%       612/ 93070     0x3f
                idle             main        ready   127    0%       160/   258     0x3f
             monitor             main    suspended   -80    0%       140/   514     0x3f
    harness report: total(3), passed(3), failed(0)
    event_suite:
    exit: test_read_write: PASSED
    exit: test_poll: PASSED
    exit: test_poll_timeout: PASSED
    
    erik@erik-PC ~/simba/simba/tst/kernel/event
    $

ivankravets, 2016-05-06:
Another unit test framework for Arduino https://github.com/mmurdoch/arduinounit

ivankravets, 2016-06-15:
Docs: http://docs.platformio.org/en/feature-platformio-30/platforms/unit_testing.html

commit bb22a129 (2016-06-15): Unit Testing for Embedded // Resolve #408
closed 2016-06-15 by ivankravets