platformio/platformio-core #412: New Framework - Simba

eerimoq, 2015-12-29:
Hello guys!

I've been working on an RTOS with a build framework, called Simba, with native, AVR and ARM toolchains, package handler, etc. However, it's useless that I provide my own toolchains and package handlers if Simba could be integrated into PlatformIO instead. If you want to have a look at Simba, see http://simba-os.readthedocs.org/en/latest/ and https://github.com/eerimoq/simba.

I'm sure you wont accept Simba as of today, because it has too few users. In fact there is only two, myself and a collegue of mine... I'm trying to make more people try it as Beta-testers, but no luck so far. PlatformIO would certainly increase the number of users, but maybe it's too early.. that's for you to decide.

And obviously I'll have to make the required changes to Simba to make it fit the PlatformIO framework.

Thanks in advance,
Erik.

itsjustvenky, 2015-12-30:
@eerimoq I will give a try. thanks.

eerimoq, 2015-12-30:
@itsjustvenky I've just started to migrate the Make based build system to `Scons`. As the first step I simply create one ``.sconscript``-file for every ``.mk``-file to keep the file structure the same. If it is the `Scons` way of structuring a build system, I don't know. It works for now.

Anyways, let's keep in touch so we don't make the same changes.

Thanks.

eerimoq, 2015-12-30:
The migration to Scons continued for ARM. All targets except the build target has been left untouched. That is, targets for test, code coverage, static code analysis, etc. I guess those should be migrated as well.

eerimoq/simba@6e320bc63d091ddbf2d371b738853a1c0f4667f0

eerimoq, 2015-12-31:
@itsjustvenky Here is the first implementation. Please have a look at the commit message and the code. There are a few problem that I need help to fix.

eerimoq/platformio@4eb5b681546e4288dcc7b494f36b1c20caad6072

commit e5d379d8 (2016-01-01): Add support for Simba framework // Resolve #412
commit f7a0374d (2016-04-28): New RTOS and build Framework named "Simba" // Resolve #412
closed 2016-04-28 by ivankravets