platformio/platformio-core #413: Depend on a specific version of a library

FWeinb, 2015-12-29:
This is probably related to #410. To support reproducible builds and work on projects that depend on the same library but a different version it would be best to support a way of installing library with the Project and not as a global package of platformio. To get this working one has to define a version range in addition to the library in [lib_install](http://docs.platformio.org/en/latest/projectconf.html#lib-install).

ivankravets, 2016-01-03:
TODO: add new `lib_requires` field to [platformio.ini](http://docs.platformio.org/en/latest/projectconf.html) with the next format:

```ini
libs_requires = 
    %LIB_NAME%: %semiver1%, %semiver2%, %semiverN%
    ID%LIB_ID%: %semiver1%, %semiver2%, %semiverN%
```

For example,
```ini
libs_requires = 
    Json: >=1.0.0, <2.0.0, !=1.5.7
    ID64: >=1.0.0, <2.0.0, !=1.5.7
```

@marvinroger, @FWeinb WDYT?

ivankravets, 2016-01-03:
Currently all libs will be installed in http://docs.platformio.org/en/latest/projectconf.html#lib-dir. What I should do when 2 projects require the same libs with different `lib_requires`? Is it good idea to install required libraries directly to project's `lib` folder?

marvinroger, 2016-01-04:
`lib_requires` should be spelled `lib_require`, as we currently have `lib_use` and `lib_ignore` without the leading `s`. So `lib_install` will be deprecated/removed? Otherwise It would sounds redundant.
And I would remove the commas, just `Json: >=1.0.0 <2.0.0 !=1.5.7` as this is more common and standard.

Maybe you could store libs by library and version. I mean:

```
* Json
    * 1.0.0
        * src...
    * 2.0.0
        * src...
```

What's the benefit of storing the library in the project folder? I can't find any. But storing it globally avoids to download multiple copies of the same library (ArduinoJson for example will probably be used in multiple projects).

marvinroger, 2016-01-04:
Plus, we would have to `.gitignore` the `lib` folder if we use Git. I'd stay on the global install.

ivankravets, 2016-01-04:
> lib_requires should be spelled lib_require, as we currently have lib_use and lib_ignore without the leading s. So lib_install will be deprecated/removed? Otherwise It would sounds redundant.
And I would remove the commas, just Json: >=1.0.0 <2.0.0 !=1.5.7 as this is more common and standard.

Done! https://github.com/platformio/platformio/issues/413#issuecomment-168556116

> Maybe you could store libs by library and version. I mean:

npm uses this scheme. Oh... How much work need to do with it. Yes, this scheme is the best from all views. Because project's lib was intended for private/specific libraries

andig, 2016-07-12:
Please make sure to add ability for requiring dev versions as well (both for libraries and platforms ;)

ivankravets, 2016-08-08:
Please re-test the latest development version.

Docs: http://docs.platformio.org/en/latest/projectconf.html#lib-deps

closed 2016-08-08 by ivankravets

andig, 2016-08-08:
I'm a little bit lost as to how `lib_deps` works with `lib_install`. Is `lib_install` obsolete now? Also, how is it possible to specify dev version for known libraries (e.g. `ESPAsyncTCP@dev`)?

Some more details for the docs would be highly appreciated.

ivankravets, 2016-08-08:
I'll write "Migration" section in the docs later.

> Is lib_install obsolete now?

`lib_install` is obsolete and is equal to `lib_deps` but need to specify ID per new line.

>  Also, how is it possible to specify dev version for known libraries (e.g. ESPAsyncTCP@dev)?

The short version for GitHub repo is

```
lib_deps =  me-no-dev/ESPAsyncTCP
```

andig, 2016-08-08:
Really off-topic now, but is it also possible to use dev version of a platform (use case is espressif here)?

ivankravets, 2016-08-08:
The same is with development platforms. They are moved to separate repositories. See https://github.com/platformio/platform-espressif

```bash
pio platform install https://github.com/platformio/platform-espressif.git
```

commit 65e0daa0 (2016-08-08): Project dependencies per build environment using "lib_deps" option // Resolve #413