platformio/platformio-core #414: Make it possible to install libraries by name

FWeinb, 2015-12-29:
It would be much easier to just install a library by name instead of a cryptic id number. And having names in the `platformio.ini` would also improve readability.

ivankravets, 2015-12-29:
I don't remember where I've written an answer for this question. The problem here is that PlatformIO Library Registry can contain more than one library with the same name. For example, it can have different `WiFi` libraries per platforms.

As solution, nee to allow to install library by name but when more than libs are found, then ask user which lib to install. Or use like `WiFi#13`, where is 100% link to lib by ID. 

Also, you have requested #410 . In this case it might look like `WiFi#13@~0.13.2`. The format will be discussed.

FWeinb, 2015-12-29:
Can it be guaranteed that the combination of platform + libraryname is unique? If that is the case, the package manager could derive the platform from the `platformio.ini` and try to search for the specific package. How is this currently handled, can a project using board=uno and board=esp01 depend on a library that only support the uno?

ivankravets, 2015-12-29:
> Can it be guaranteed that the combination of platform + libraryname is unique?

This is not PlatformIO guaranty. The owner of the lib specifies which frameworks and platforms are supported. Currently, there are no any checks for compatibility. PlatformIO builds all that user asks. See #415

jasonmhite, 2016-01-26:
> The problem here is that PlatformIO Library Registry can contain more than one library with the same name. For example, it can have different WiFi libraries per platforms.

Wouldn't it make sense to ask for a particular selection if there are multiple options that match a given name? Something like "there are multiple libraries that match the name <foolib>, choes one of: 1. foo1 2. foo2 3. foo3"? Similarly, when building with `platformio run`, if it runs into a conflict then it can pause and ask for the proper choice and simply remember that in the future, perhaps by writing to `platformio.ini`.

ivankravets, 2016-01-26:
@jasonmhite this is `feature` issue, I'll implement it.

commit 758396c9 (2016-08-02): PlatformIO Library Manager 3.0 // Resolve #588 , Resolve #414, Resolve #498, Resolve #475, Resolve #410, Resolve #461, Resolve #361, Resolve #507, Resolve #554
closed 2016-08-08 by ivankravets