platformio/platformio-core #417: Problems with esp8266 OTA

igorjan, 2015-12-30:
When I  try the OTA example found here:
https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino

I encounterd when uploading the OTA script didn’t seem to recognize my input IP. I was getting:
14:21:38 [CRITICAL]: Not enough arguments.

It seems on windows you must specify upload port in ” ” like this.
upload_port = “192.168.1.40”

When I set the IP in " " I  get this output when I do an upload:
esptool v0.4.6 – (c) 2014 Ch. Klippel
setting board to nodemcu
setting baudrate from 115200 to 115200
setting port from COM1 to 192.168.1.40
espcomm_upload_file
stat .pioenvs\nodemcu\firmware.bin success
error: Failed to open 192.168.1.40
error: espcomm_open failed
scons: *** [upload] Error -1
[ ERROR ]

Any Idea? I disabled the firewall and the IP is definitely OK. I tested the UDP port 8266 and it is open.
Also there is no activity on the serial monitor when I try OTA.

I think the problem is still in the IP address somehow it is not recognized properly.

itsjustvenky, 2015-12-30:
@igorjan  Can you try testing "ESPWebServer/HelloSever.ino" (inarduino ide examples) and see if you can access the mentioned IP-Address and also try to ping the ip-address and see if you get any response.

igorjan, 2015-12-30:
@itsjustvenky the HelloServer.ino works without a problem. I get "hello from esp8266!" in the web borwser.

I can ping the IP, also I tested the UDP port 8266 on the ESP which is used for OTA and it seems to be open.

itsjustvenky, 2015-12-30:
@igorjan  which version of platformio are you using ?

igorjan, 2015-12-30:
@itsjustvenky platformio version 2.6.3

ivankravets, 2015-12-30:
@igorjan try to specify IP as http://docs.platformio.org/en/latest/projectconf.html#upload-port . Don't pass it to console as argument.

igorjan, 2015-12-30:
Here you can see 3 attempts with difrent definitions of upload-port.

**Attempt 1:**
C:\Igor_i\ESP8266\Project1>platformio run --target upload --upload-port 192.168.1.40
[12/30/15 16:19:27] Processing nodemcu (platform: espressif, board: nodemcu, framework: arduino)
-------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs\nodemcu\firmware.bin"])
"C:\Users\igor\.platformio\packages\framework-arduinoespressif\tools\espota.py"
--debug --progress -i 192.168.1.40 -f .pioenvs\nodemcu\firmware.bin
16:19:31 [CRITICAL]: Not enough arguments.
scons: *** [upload] Error 1
========================= [ ERROR ] Took 3.44 seconds =========================

**Attempt 2:**
C:\Igor_i\ESP8266\Project1>platformio run --target upload --upload-port "192.168.1.40"
[12/30/15 16:19:47] Processing nodemcu (platform: espressif, board: nodemcu, framework: arduino)
-------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs\nodemcu\firmware.bin"])
"C:\Users\igor\.platformio\packages\framework-arduinoespressif\tools\espota.py"
--debug --progress -i 192.168.1.40 -f .pioenvs\nodemcu\firmware.bin
16:19:49 [CRITICAL]: Not enough arguments.
scons: *** [upload] Error 1
========================= [ ERROR ] Took 2.63 seconds =========================

**Atempt 3**
C:\Igor_i\ESP8266\Project1>platformio run --target upload --upload-port '192.168.1.40'
[12/30/15 16:20:16] Processing nodemcu (platform: espressif, board: nodemcu, framework: arduino)
-------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs\nodemcu\firmware.bin"])
"C:\Users\igor\.platformio\packages\tool-esptool\esptool" -vv -cd nodemcu -cb 11
5200 -cp '192.168.1.40' -cf .pioenvs\nodemcu\firmware.bin
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to nodemcu
setting baudrate from 115200 to 115200
setting port from COM1 to '192.168.1.40'
espcomm_upload_files
cons: *** [upload] Error -1stat .pioenvs\nodemcu\firmware.bin success

error: Failed to open '192.168.1.40'
error: espcomm_open failed
========================= [ ERROR ] Took 2.46 seconds =========================

ivankravets, 2015-12-30:
Please read my comment above https://github.com/platformio/platformio/issues/417#issuecomment-168017241. Don't pass IP to terminal. See example http://docs.platformio.org/en/latest/platforms/espressif.html#over-the-air-ota-update

Add this line to your `platformio.ini`
```
upload_port = 192.168.1.40
```

skorokithakis, 2015-12-30:
Isn't this a windows/click/esptool problem? If windows tokenizes periods as separate tokens, you need to quote it if you want to pass it in the command line.

igorjan, 2015-12-30:
@ivankravets sorry I misunderstood you. I was doing it with platformio.ini file all along.

ivankravets, 2015-12-30:
@igorjan please wait for the new development version. I'll ask you to test it. 1-2mins

ivankravets, 2015-12-30:
@igorjan could you paste directly yo terminal this
```
"C:\Users\igor.platformio\packages\tool-esptool\esptool" -vv -cd nodemcu -cb 11
5200 -cp "192.168.1.40" -cf .pioenvs\nodemcu\firmware.bin
```

Does it work?

igorjan, 2015-12-30:
@ivankravets  
c:\Users\igor\.platformio\packages\tool-esptool>esptool.exe -vv -cd nodemcu -cb
115200 -cp "192.168.1.40" -cf C:\Igor_i\ESP8266\Project1\.pioenvs\nodemcu\firmwa
re.bin
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting board to nodemcu
        setting baudrate from 115200 to 115200
        setting port from COM1 to 192.168.1.40
        espcomm_upload_file
        stat C:\Igor_i\ESP8266\Project1\.pioenvs\nodemcu\firmware.bin success
error: Failed to open 192.168.1.40
error: espcomm_open failed

ivankravets, 2015-12-30:
@igorjan STOP STOP :no_good: 

Did you upload OTA sketch before make OTA uploading?

ivankravets, 2015-12-30:
No here problems with IP. Looks like you didn't upload OTA sketch and din't configured it.

itsjustvenky, 2015-12-30:
damn... we totally forgot the basic question...

ivankravets, 2015-12-30:
@itsjustvenky :) :cocktail: New Year...

namoaton, 2015-12-30:
[esp_OTA.zip](https://github.com/platformio/platformio/files/74949/esp_OTA.zip)

igorjan, 2015-12-30:
Yes I did have the OTA example uploaded via serial before trying OTA. I even changed some strings in the serial.print so I'm sure that the correct sketch is uploaded.

ivankravets, 2015-12-30:
@igorjan could you make OTA uploading using Arduino IDE? See https://github.com/esp8266/Arduino/blob/master/doc/filesystem.md#uploading-files-to-file-system

igorjan, 2015-12-30:
@namoaton tried your project it is the same

igorjan, 2015-12-30:
@ivankravets Will try arduino but have to install it first

itsjustvenky, 2015-12-30:
@igorjan https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md

Try this arduino build : Arduino-PR-4107-BUILD-421

else you need to manually set the IP-Address in arduino preference files.

itsjustvenky, 2015-12-30:
https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md#arduinoota

igorjan, 2015-12-30:
OK tried the arduino and it works without a problem. The sketch is the same.

ivankravets, 2015-12-30:
@igorjan print here `platformio --version``

igorjan, 2015-12-30:
c:\Users\User\.platformio\packages\tool-esptool>platformio --version
PlatformIO, version 2.6.3

ivankravets, 2015-12-30:
@igorjan please install the latest development version http://docs.platformio.org/en/latest/installation.html#d-development-version

I've just made a few changes for you.

igorjan, 2015-12-30:
Now I have: PlatformIO, version 2.7.0.dev12

But the problem is the same

"C:\Users\igor\.platformio\packages\tool-esptool\esptool" -vv -cd nodemcu -cb 115200 -cp "192.168.1.40" -cf .pioenvs\nodemcu\firmware.bin
esptool v0.4.6 - (c) 2014 Ch. Klippel 
setting board to nodemcu
setting baudrate from 115200 to 115200
setting port from COM1 to 192.168.1.40
espcomm_upload_file
stat .pioenvs\nodemcu\firmware.bin success
error: Failed to open 192.168.1.40
error: espcomm_open failed
scons: *** [upload] Error -1
 [ ERROR ] Took 11.14 seconds 
[Finished in 11.882s]

ivankravets, 2015-12-30:
@igorjan I've jot this error. Sorry, hard day is today. :sad:

Please wait for the new build

ivankravets, 2015-12-30:
@igorjan try to reinstall the latest PlatformIO 2.7.0.dev13

itsjustvenky, 2015-12-30:
esptool doesn't do OTA, its only for  local ports and not ip-address.

espota.py is the tool we need to try.

 python espota.py -i 192.168.1.40 -p8266 -f .pioenvs\nodemcu\firmware.bin

itsjustvenky, 2015-12-30:
@igorjan  can you give us the exact command you are trying ?

igorjan, 2015-12-30:
C:\Igor_i\ESP8266\Project1>platformio run --target upload

[12/30/15 18:06:01] Processing nodemcu (platform: espressif, upload_port: "192.168.1.40", board: nodemcu, framework: arduino)
-------------------------------------------------------------------------------
<lambda>(["upload"], [".pioenvs\nodemcu\firmware.bin"])
"C:\Users\igor\.platformio\packages\tool-esptool\esptool" -vv -cd nodemcu -cb 11
5200 -cp "192.168.1.40" -cf .pioenvs\nodemcu\firmware.bin
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to nodemcu
setting baudrate from 115200 to 115200
setting port from COM1 to 192.168.1.40
secons: *** [upload] Error -1spcomm_upload_file

stat .pioenvs\nodemcu\firmware.bin success
error: Failed to open 192.168.1.40
error: espcomm_open failed
========================= [ ERROR ] Took 2.31 seconds =========================

ivankravets, 2015-12-30:
@igorjan please remove double quotes

igorjan, 2015-12-30:
Thus command works without a problem

python espota.py -i 192.168.1.40 -p8266 -f .pioenvs\nodemcu\firmware.bin

ivankravets, 2015-12-30:
Please install again the latest PlatformIO.

itsjustvenky, 2015-12-30:
@igorjan 

the actual command is

```
platformio run --target upload --upload-port 192.168.1.40

igorjan, 2015-12-30:
@itsjustvenky for your command I get:
C:\Igor_i\ESP8266\Project1>platformio run --target upload --upload-port 192.168.
1.40
[12/30/15 18:13:43] Processing nodemcu (platform: espressif, upload_port: 192.16
8.1.40, board: nodemcu, framework: arduino)
-------------------------------------------------------------------------------
<lambda>(["upload"], [".pioenvs\nodemcu\firmware.bin"])
"C:\Users\igor\.platformio\packages\framework-arduinoespressif\tools\espota.py"
--debug --progress -i 192.168.1.40 -f .pioenvs\nodemcu\firmware.bin
18:13:45 [CRITICAL]: Not enough arguments.
scons: *** [upload] Error 1

igorjan, 2015-12-30:
@ivankravets if I remove the double qutes. I get the error:

18:13:45 [CRITICAL]: Not enough arguments.
scons: *** [upload] Error 1

The command python espota.py -i 192.168.1.40 -p8266 -f .pioenvs\nodemcu\firmware.bin
works without a problem.

ivankravets, 2015-12-30:
Ok, now please revert `upload_port: "192.168.1.40"` in `platformio.ini`.

itsjustvenky, 2015-12-30:
@igorjan  I think "ivankravets" just fixed something related this ip-address. Please install latest dev. version.

ivankravets, 2015-12-30:
I don't understand behaviour of your machine. Please revert back quotes. If it helps, I'll added them to code.

ivankravets, 2015-12-30:
I don't see difference between `"C:\Users\igor.platformio\packages\framework-arduinoespressif\tools\espota.py" --debug --progress -i 192.168.1.40 -f .pioenvs\nodemcu\firmware.bin` and `espota.py -i 192.168.1.40 -p8266 -f .pioenvs\nodemcu\firmware.bin`

ivankravets, 2015-12-30:
@igorjan Could you try without TCP/IP port? `espota.py -i 192.168.1.40 -f .pioenvs\nodemcu\firmware.bin`. Does it work?

skorokithakis, 2015-12-30:
@ivankravets notice that he's running esptool, not espota. Why would that work?

~~~
"C:\Users\igor.platformio\packages\tool-esptool\esptool" -vv -cd nodemcu -cb 115200 -cp "192.168.1.40" -cf .pioenvs\nodemcu\firmware.bin
~~~

igorjan, 2015-12-30:
OK this works too:

python espota.py -i 192.168.1.40 -f C:\Igor_i\ESP8266\Project1\.pioenvs\nodemcu\firmware.bin

igorjan, 2015-12-30:
Also in version dev13 I get error 

18:18:56 [CRITICAL]: Not enough arguments.

With or without qutes for upload port

ivankravets, 2015-12-30:
I've just pushed dev14. Please try it

igorjan, 2015-12-30:
Dev 14 is the same

18:18:56 [CRITICAL]: Not enough arguments.
With or without qutes for upload port 

If you guys want I can give you team viewer access to the computer so you can test yourself

skorokithakis, 2015-12-30:
@igorjan Can you paste your platformio.ini here?

igorjan, 2015-12-30:
#
# Project Configuration File
#
# A detailed documentation with the EXAMPLES is located here:
# http://docs.platformio.org/en/latest/projectconf.html
#

# A sign `#` at the beginning of the line indicates a comment
# Comment lines are ignored.

# Simple and base environment
# [env:mybaseenv]
# platform = %INSTALLED_PLATFORM_NAME_HERE%
# framework =
# board =
#
# Automatic targets - enable auto-uploading
# targets = upload

[env:nodemcu]
platform = espressif
framework = arduino
board = nodemcu
upload_port = "192.168.1.40"

ivankravets, 2015-12-30:
Yes, please mail me@ikravets.com with TeamViewer Id/pass. Thanks

itsjustvenky, 2015-12-30:
@igorjan can you post all the text from your console from the starting of platformio run...

skorokithakis, 2015-12-30:
@igorjan You should surround your code with three tildes:

```
~~~
like this
~~~
```

skorokithakis, 2015-12-30:
This seems to fix it:

~~~
        upload_port = env.get("UPLOAD_PORT", "").strip("\"\' ")
        if (upload_port and (
                upload_port.endswith(".local") or
                socket.inet_aton(upload_port))):
            env.Replace(
                UPLOADCMD="$UPLOADOTACMD"
            )
~~~

ivankravets, 2015-12-30:
Guys, the problem was linked with launching `espota.py`. On this machine PyCharm IDE handles all Python files by default.

I'll fix it and paste directly current Python interpreter

ivankravets, 2015-12-30:
@igorjan please try the latest `dev15`

closed 2015-12-30 by ivankravets

igorjan, 2015-12-30:
:) This one works without problem

Thank you guys for your help, really appreciate it.

ivankravets, 2015-12-30:
Thank you too that report it. I'm going to release 2.7.0 :christmas_tree:  :blush:

ivankravets, 2015-12-30:
@itsjustvenky, @skorokithakis and @namoaton  - THANKS A LOT FOR THE HELP! :beers:

[1 more reply]

commit 8d6aee31 (2015-12-30): Use current Python interpreter for Python-based tools // Resolve #417