platformio/platformio-core #421: CMakeLists.txt add_executable has only one source file

hameno, 2016-01-01:
Hello,
I'm not sure if this is a bug, missing feature oder just expected but when I run 
```platformio init --board uni --ide clion``` 
only one cpp file from my src folder is added to the ```add_executable``` setting.
I would expect all cpp files to be added there.

ivankravets, 2016-01-01:
This is OK. CLion uses Cmake as core build system and depends on it. However, PlatformIO has own build system and doesn't depend on Cmake. PlatformIO generates minimal settings in `CMakeLists.txt` which are required to code autocompletion and etc.

hameno, 2016-01-01:
But without them code completion in clion is not completely available for all files.

ivankravets, 2016-01-01:
Sure? Could you provide me simple projects with more that 1 file where I will see problems with autocompletion and etc.

Thanks.

hameno, 2016-01-01:
Sure, but it is not exactly code completion thats not working but rather the code insights on the side of the code (where the line numbers are). 
My test case consists of the main.cpp and one class with header and source. When not added to the CmakeLists.txt the header file will not show links to the implemented methods in the source.
This happens with latest stable of platformio (2.7.0) and clion (1.2.4).

With SampleClass.cpp added to add_executable:
[image]
and without:
[image]
[ClionSample.zip](https://github.com/platformio/platformio/files/75948/ClionSample.zip)

ivankravets, 2016-01-01:
Thanks, will fix it.

ivankravets, 2016-01-04:
@hameno please re-test it with the latest http://docs.platformio.org/en/latest/installation.html#d-development-version

closed 2016-01-04 by ivankravets

hameno, 2016-01-06:
Looks good!

diejmon, 2017-02-05:
What about files in ./lib folder? Code insights doesn't work in this folder.
_CLion 2016.3.2_
_PlatformIO, version 3.3.0a10_

ivankravets, 2017-02-05:
@diejmon have you tried to re-init project as described in docs?
* http://docs.platformio.org/en/latest/ide/clion.html

diejmon, 2017-02-05:
@ivankravets yes, just tried this again. If I add all files in ./lib folder to `add_executable(${PROJECT_NAME} ${SRC_LIST})` list, CLion works as expected.

ivankravets, 2017-02-05:
@diejmon I've got you. Please file a new issue https://github.com/platformio/platformio-core

Thanks!

diejmon, 2017-02-05:
@ivankravets Thanks for fast response! Will create a new issue soon.

diejmon, 2017-02-05:
#897

commit 0751a2ce (2016-01-04): Fix CMakeLists.txt add_executable has only one source file // Resolve #421