platformio/platformio-core #422: platformio init failes when --ide used

nigelb, 2016-01-01:
As per the [IDE Integration](http://docs.platformio.org/en/latest/ide.html) page:
```shell
$> platformio init --ide clion --board uno

The current working directory /home/user/tmp/clion_platform_test will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories will be created in /home/user/tmp/clion_platform_test
platformio.ini - Project Configuration File. |-> PLEASE EDIT ME <-|
src - Put your source files here
lib - Put here project specific (private) libraries
Do you want to continue? [y/N]: y
Error: Traceback (most recent call last):
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio-2.7.0-py2.7.egg/platformio/__main__.py", line 97, in main
    cli(None, None, None)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio-2.7.0-py2.7.egg/platformio/commands/init.py", line 113, in cli
    project_dir, ide, board[0])
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio-2.7.0-py2.7.egg/platformio/ide/projectgenerator.py", line 34, in __init__
    self._gather_tplvars()
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio-2.7.0-py2.7.egg/platformio/ide/projectgenerator.py", line 146, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio-2.7.0-py2.7.egg/platformio/util.py", line 102, in __call__
    value = self.func(*args)
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio-2.7.0-py2.7.egg/platformio/ide/projectgenerator.py", line 77, in get_project_build_data
    start_index = output.index('\n{"')
ValueError: substring not found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================
```
Verify that I have the latest version of PlatformIO:

```shell
$> ~/.python/platform/bin/pip install -U platformio
Requirement already up-to-date: platformio in /home/user/.python/platform/lib/python2.7/site-packages/platformio-2.7.0-py2.7.egg
Requirement already up-to-date: bottle<0.13 in /home/user/.python/platform/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: click>=3.2,<6 in /home/user/.python/platform/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: lockfile>=0.9.1,<0.13 in /home/user/.python/platform/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: requests>=2.4.0,<3 in /home/user/.python/platform/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: colorama in /home/user/.python/platform/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: pyserial<4 in /home/user/.python/platform/lib/python2.7/site-packages (from platformio)
Cleaning up...
```
I even install from the head of the repository and the problem persists.

ivankravets, 2016-01-01:
1. Please revert back to the stable version. Don't use HEAD
```
pip uninstall platformio
pip install platformio
```
2. Looks like you have problems with installed packages. Give me please an output of `platformio platforms list`

nigelb, 2016-01-01:
I have reverted back to stable and run:
```shell
$> platformio platforms list
atmelavr    with packages: toolchain-atmelavr, tool-avrdude, framework-arduinoavr
```

ivankravets, 2016-01-01:
Could you try to build Blink project? https://github.com/platformio/platformio/tree/develop/examples/wiring-blink

nigelb, 2016-01-01:
It compiles fine, and it installed a bunch of platforms:
```shell
$> platformio platforms list               
atmelavr    with packages: toolchain-atmelavr, tool-avrdude, framework-arduinoavr
espressif   with packages: toolchain-xtensa, tool-esptool, framework-arduinoespressif, ldscripts
teensy      with packages: framework-arduinoteensy, toolchain-gccarmnoneeabi, toolchain-atmelavr, ldscripts
timsp430    with packages: toolchain-timsp430, framework-energiamsp430
```

running `platformio init --ide clion --board uno` still fails though.

ivankravets, 2016-01-01:
This is good. Now please give me an output of `platformio run -t idedata`

nigelb, 2016-01-01:
```shell
$> platformio run -t idedata        
[Sat Jan  2 08:22:31 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
{"includes": ["/home/user/.platformio/packages/framework-arduinoavr/variants/standard", "/home/user/.platformio/packages/framework-arduinoavr/cores/arduino", "/home/user/.platformio/packages/toolchain-atmelavr/avr/include", "/home/user/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/4.8.1/include-fixed", "/home/user/.platformio/packages/toolchain-atmelavr/lib/gcc/avr/4.8.1/include"], "cxx_path": "/home/user/.platformio/packages/toolchain-atmelavr/bin/avr-g++", "defines": ["F_CPU=16000000L", "ARDUINO_ARCH_AVR", "ARDUINO_AVR_UNO", "ARDUINO=10607", "PLATFORMIO=020700", "__AVR_ATmega328P__"]}
============================================================================= [SUCCESS] Took 0.20 seconds =============================================================================

[Sat Jan  2 08:22:31 2016] Processing nodemcu (platform: espressif, board: nodemcu, framework: arduino, build_flags: -D LED_BUILTIN=BUILTIN_LED)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
{"includes": ["/home/user/.platformio/packages/framework-arduinoespressif/variants/nodemcu", "/home/user/.platformio/packages/framework-arduinoespressif/cores/esp8266", "/home/user/.platformio/packages/framework-arduinoespressif/cores/esp8266/spiffs", "/home/user/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include", "/home/user/.platformio/packages/toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/include-fixed", "/home/user/.platformio/packages/toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/include"], "cxx_path": "/home/user/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-g++", "defines": ["F_CPU=80000000L", "__ets__", "ICACHE_FLASH", "ESP8266", "ARDUINO_ARCH_ESP8266", "ARDUINO_ESP8266_ESP12", "LED_BUILTIN=BUILTIN_LED", "ARDUINO=200", "PLATFORMIO=020700"]}
============================================================================= [SUCCESS] Took 0.21 seconds =============================================================================

[Sat Jan  2 08:22:31 2016] Processing teensy31 (platform: teensy, board: teensy31, framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
{"includes": ["/home/user/.platformio/packages/framework-arduinoteensy/cores/teensy3", "/home/user/.platformio/packages/framework-arduinoteensy/cores/teensy3/avr", "/home/user/.platformio/packages/framework-arduinoteensy/cores/teensy3/util", "/home/user/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include", "/home/user/.platformio/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/4.8.4/include-fixed", "/home/user/.platformio/packages/toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/4.8.4/include"], "cxx_path": "/home/user/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-g++", "defines": ["F_CPU=72000000L", "USB_SERIAL", "LAYOUT_US_ENGLISH", "__MK20DX256__", "ARDUINO=10600", "TEENSYDUINO=124", "PLATFORMIO=020700"]}
============================================================================= [SUCCESS] Took 0.20 seconds =============================================================================

[Sat Jan  2 08:22:31 2016] Processing lpmsp430g2553 (platform: timsp430, board: lpmsp430g2553, framework: energia, build_flags: -D LED_BUILTIN=RED_LED)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
{"includes": ["/home/user/.platformio/packages/framework-energiamsp430/variants/launchpad", "/home/user/.platformio/packages/framework-energiamsp430/cores/msp430", "/home/user/.platformio/packages/framework-energiamsp430/cores/msp430/avr", "/home/user/.platformio/packages/toolchain-timsp430/msp430/include", "/home/user/.platformio/packages/toolchain-timsp430/lib/gcc/msp430/4.6.3/include-fixed", "/home/user/.platformio/packages/toolchain-timsp430/lib/gcc/msp430/4.6.3/include"], "cxx_path": "/home/user/.platformio/packages/toolchain-timsp430/bin/msp430-g++", "defines": ["F_CPU=16000000L", "LED_BUILTIN=RED_LED", "ARDUINO=101", "ENERGIA=15", "PLATFORMIO=020700"]}
============================================================================= [SUCCESS] Took 0.18 seconds =============================================================================
```

ivankravets, 2016-01-01:
Do you run `platformio init --ide clion --board uno` in empty folder? Or in existing project?

ivankravets, 2016-01-01:
See my log when I init project in empty directory
```
platformio init -d /tmp/aaa/ --ide clion --board uno
The next files/directories will be created in /private/tmp/aaa
platformio.ini - Project Configuration File. |-> PLEASE EDIT ME <-|
src - Put your source files here
lib - Put here project specific (private) libraries
Do you want to continue? [y/N]: y
Warning! Can not find main source file (*.c, *.cpp). So, code auto-completion is disabled. Please add source files to `src` directory and re-initialize project or edit `CMakeLists.txt` file manually (`add_executable` command).

Project has been successfully initialized!
Useful commands:
`platformio run` - process/build project from the current directory
`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
`platformio run --target clean` - clean project (remove compiled files)
`platformio run --help` - additional information
```

nigelb, 2016-01-01:
Both
```shell
$> platformio init -d clion_test_empty --ide clion --board uno
The next files/directories will be created in /home/user/tmp/clion_test_empty
platformio.ini - Project Configuration File. |-> PLEASE EDIT ME <-|
src - Put your source files here
lib - Put here project specific (private) libraries
Do you want to continue? [y/N]: y
Warning! Can not find main source file (*.c, *.cpp). So, code auto-completion is disabled. Please add source files to `src` directory and re-initialize project or edit `CMakeLists.txt` file manually (`add_executable` command).
Error: Traceback (most recent call last):
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/__main__.py", line 97, in main
    cli(None, None, None)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/commands/init.py", line 113, in cli
    project_dir, ide, board[0])
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 34, in __init__
    self._gather_tplvars()
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 146, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/util.py", line 102, in __call__
    value = self.func(*args)
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 77, in get_project_build_data
    start_index = output.index('\n{"')
ValueError: substring not found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================
```

```shell
$> platformio init --ide clion --board uno

The current working directory /home/user/tmp/clion_test_not_empty will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories will be created in /home/user/tmp/clion_test_not_empty
platformio.ini - Project Configuration File. |-> PLEASE EDIT ME <-|
src - Put your source files here
lib - Put here project specific (private) libraries
Do you want to continue? [y/N]: y
Error: Traceback (most recent call last):
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/__main__.py", line 97, in main
    cli(None, None, None)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/commands/init.py", line 113, in cli
    project_dir, ide, board[0])
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 34, in __init__
    self._gather_tplvars()
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 146, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/util.py", line 102, in __call__
    value = self.func(*args)
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 77, in get_project_build_data
    start_index = output.index('\n{"')
ValueError: substring not found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================
```

ivankravets, 2016-01-01:
Please try PlatformIO 2.7.1.dev0 http://docs.platformio.org/en/latest/installation.html#d-development-version

nigelb, 2016-01-02:
A different error now:

```shell
$> pip uninstall platformio

Uninstalling platformio:
  /home/user/.python/platform/bin/platformio
  /home/user/.python/platform/lib/python2.7/site-packages/platformio-2.7.1.dev0-py2.7.egg
Proceed (y/n)? y
  Successfully uninstalled platformio

$> pip install -U https://github.com/platformio/platformio/archive/develop.zip
Downloading/unpacking https://github.com/platformio/platformio/archive/develop.zip
  Downloading develop.zip (8.9MB): 8.9MB downloaded
  Running setup.py egg_info for package from https://github.com/platformio/platformio/archive/develop.zip
    
Requirement already up-to-date: bottle<0.13 in ./lib/python2.7/site-packages (from platformio==2.7.1.dev0)
Requirement already up-to-date: click>=3.2,<6 in ./lib/python2.7/site-packages (from platformio==2.7.1.dev0)
Requirement already up-to-date: lockfile>=0.9.1,<0.13 in ./lib/python2.7/site-packages (from platformio==2.7.1.dev0)
Requirement already up-to-date: requests>=2.4.0,<3 in ./lib/python2.7/site-packages (from platformio==2.7.1.dev0)
Requirement already up-to-date: colorama in ./lib/python2.7/site-packages (from platformio==2.7.1.dev0)
Requirement already up-to-date: pyserial<4 in ./lib/python2.7/site-packages (from platformio==2.7.1.dev0)
Installing collected packages: platformio
  Found existing installation: platformio 2.7.0
    Uninstalling platformio:
      Successfully uninstalled platformio
  Running setup.py install for platformio
    
    Installing platformio script to /home/user/.python/platform/bin
Successfully installed platformio
Cleaning up...

$> platformio init --ide clion --board uno

The current working directory /home/user/tmp/clion_test_empty will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories will be created in /home/user/tmp/clion_test_empty
platformio.ini - Project Configuration File. |-> PLEASE EDIT ME <-|
src - Put your source files here
lib - Put here project specific (private) libraries
Do you want to continue? [y/N]: y
Warning! Can not find main source file (*.c, *.cpp). So, code auto-completion is disabled. Please add source files to `src` directory and re-initialize project or edit `CMakeLists.txt` file manually (`add_executable` command).
Error: Traceback (most recent call last):
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/__main__.py", line 97, in main
    cli(None, None, None)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/commands/init.py", line 113, in cli
    project_dir, ide, board[0])
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 34, in __init__
    self._gather_tplvars()
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 146, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/util.py", line 102, in __call__
    value = self.func(*args)
  File "/home/user/.python/platform/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 79, in get_project_build_data
    data = json.loads(output[start_index + 1:stop_index + 1])
  File "/usr/lib64/python2.7/json/__init__.py", line 338, in loads
    return _default_decoder.decode(s)
  File "/usr/lib64/python2.7/json/decoder.py", line 368, in decode
    raise ValueError(errmsg("Extra data", s, end, len(s)))
ValueError: Extra data: line 1 column 11 - line 1 column 599 (char 10 - 598)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================
```

ivankravets, 2016-01-02:
Please re-test latest dev build.

P.S: I still don't understand why doesn't exists `\n` before `{` in your configuration. The results above show that you have new line before `{`.

closed 2016-01-02 by ivankravets

nigelb, 2016-01-06:
Worked fine. Thanks.

commit b1de8d11 (2016-01-02): Fix project generator for CLion IDE // Resolve #422
commit 7ca5ef7e (2016-01-02): Merge branch 'develop' into feature/platformio-30

* develop:
  Fix project generator for CLion IDE // Resolve #422
  Fix project generator for CLion IDE // Issue #422