platformio/platformio-core #423: PlatformIO 3.0 Roadmap

ivankravets, 2016-01-01:
# [Milestone 3.0](https://github.com/platformio/platformio/milestones/3.0.0)
- [ ] Handle specific extra build flags from library.json #298 
- [ ] Add openOCD support #341
- [ ] Allow GDB protocol for ARM-based platforms #342
- [ ] Arduino ZERO as board type? #356
- [ ] Unit testing for embedded #408 
- [ ] Support versioning librarys via semver #410
- [ ] New Framework - Simba #412
- [ ] Depend on a specific version of a library #413
- [ ] Make it possible to install libraries by name #414
- [ ] Check library compatibility with build environment #415

# Install the latest PlatformIO 3.0 development version
```
pip uninstall platformio
pip install https://github.com/platformio/platformio/archive/feature/platformio-30.zip
```
P.S: To be actual with the latest development PlatformIO 3.0 you will need to repeat commands above each new commit 

# Revert back to stable PlatformIO 2.0
```
pip uninstall platformio
pip install platformio
```

Obijuan, 2016-02-05:
I've already written the code for an initial support for the Lattice ice40 FPGA new platform (issue #480) I would like to pull-request it (for Platformio 3.0)

My questions are:
1)  Can I proceed? or should I wait for something
2) In case I can proceed. Should I create a branch of develop? or of feature/platformio-30? or of another?

Thanks a lot for your work

Obijuan, 2016-02-23:
Just waiting for an answer on how to proceed. If no answer i will create a branch from develop an emitt the pull request

ivankravets, 2016-02-23:
@Obijuan PlatformIO 3.0 will not contain `builder/scripts` folder with platform. All platforms will live in separate repositories. 

Please wait for PlatformIO 3.0. I will report here further steps.

Thanks.

Obijuan, 2016-02-23:
ok. Thanks

closed 2016-10-27 by ivankravets