platformio/platformio-core #429: Unable to build a project on MacOS 10.11.2

neverov, 2016-01-03:
I get python error when trying to build a simple one-file solution:

```
> platformio run
[Sun Jan  3 16:01:34 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------
The platform 'atmelavr' has not been installed yet. Would you like to install it now? [y/N]: y
Installing toolchain-atmelavr package:
Downloading  [####################################]  100%          
Error: Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/platformio/__main__.py", line 97, in main
    cli(None, None, None)
  File "/Library/Python/2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Library/Python/2.7/site-packages/platformio/commands/run.py", line 85, in cli
    results.append(ep.process())
  File "/Library/Python/2.7/site-packages/platformio/commands/run.py", line 121, in process
    result = self._run()
  File "/Library/Python/2.7/site-packages/platformio/commands/run.py", line 187, in _run
    return p.run(build_vars, build_targets, self.verbose_level)
  File "/Library/Python/2.7/site-packages/platformio/platforms/base.py", line 423, in run
    self._install_default_packages()
  File "/Library/Python/2.7/site-packages/platformio/platforms/base.py", line 409, in _install_default_packages
    return self.install()
  File "/Library/Python/2.7/site-packages/platformio/platforms/base.py", line 326, in install
    pm.install(name)
  File "/Library/Python/2.7/site-packages/platformio/pkgmanager.py", line 102, in install
    dlpath = self.download(info['url'], pkg_dir, info['sha1'])
  File "/Library/Python/2.7/site-packages/platformio/pkgmanager.py", line 46, in download
    fd.verify(sha1)
  File "/Library/Python/2.7/site-packages/platformio/downloader.py", line 89, in verify
    result = util.exec_command(["sha1sum", self._destination])
  File "/Library/Python/2.7/site-packages/platformio/util.py", line 242, in exec_command
    p = subprocess.Popen(*args, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1334, in _execute_child
    child_exception = pickle.loads(data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 1382, in loads
    return Unpickler(file).load()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 858, in load
    dispatch[key](self)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 966, in load_string
    raise ValueError, "insecure string pickle"
ValueError: insecure string pickle

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================
```

```
> cat platformio.ini 
#
# Project Configuration File
#
# A detailed documentation with the EXAMPLES is located here:
# http://docs.platformio.org/en/latest/projectconf.html
#

# A sign `#` at the beginning of the line indicates a comment
# Comment lines are ignored.

# Simple and base environment
# [env:mybaseenv]
# platform = %INSTALLED_PLATFORM_NAME_HERE%
# framework =
# board =
#
# Automatic targets - enable auto-uploading
# targets = upload

[env:uno]
platform = atmelavr
framework = arduino
board = uno
```

```
> sw_vers
ProductName:	Mac OS X
ProductVersion:	10.11.2
BuildVersion:	15C50
```

ivankravets, 2016-01-03:
Wow! I don't know what I should to do here http://stackoverflow.com/questions/1746825/python-valueerror-insecure-string-pickle

> are much more likely than a never-observed bug in Python itself in a functionality that's used billions of times a day all over the world

:smile: 

Looks like internal Python's problem. Could try install platform manually?

```
platformio platforms install atmelavr
```

neverov, 2016-01-03:
Hmm, definitely looks like a python problem. 
I tried with python out-of the box `2.7.10` and python from `homebrew`, which is `2.7.11` and the result persists. :(

Which version do you use?

ivankravets, 2016-01-03:
I've never seen this error. I use default Python in Mac OS X

> python --version
Python 2.7.10

Could you try
```
> shasum /usr/bin/shasum
```

?

neverov, 2016-01-03:
Gives me: `dcb56150088c64188cfcbf84cb0fc0b83f00a7a1  /usr/bin/shasum5.18`.

I guess the problem is in my python setup, I'll try to play with `virtualenv` tomorrow and figure out what's going wrong.

ivankravets, 2016-01-03:
Test this please
```
> python
Python 2.7.10 (default, Oct 23 2015, 18:05:06)
[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import subprocess
>>> subprocess.check_output(["shasum", "/usr/bin/shasum"])
'dcb56150088c64188cfcbf84cb0fc0b83f00a7a1  /usr/bin/shasum5.18\n'
>>>
```

neverov, 2016-01-03:
Yep, exactly the same result
```
> python
Python 2.7.10 (default, Oct 23 2015, 18:05:06) 
[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import subprocess
>>> subprocess.check_output(["shasum", "/usr/bin/shasum"])
'dcb56150088c64188cfcbf84cb0fc0b83f00a7a1  /usr/bin/shasum5.18\n'
>>> 

```

ivankravets, 2016-01-03:
Now it looks interesting... And
```
> python
Python 2.7.10 (default, Oct 23 2015, 18:05:06) 
[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.5)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import subprocess
>>> subprocess.check_output(["sha1sum", "/usr/bin/shasum"])
```

neverov, 2016-01-03:
Here we go:
```python
>>> import subprocess
>>> subprocess.check_output(["sha1sum", "/usr/bin/shasum"])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 566, in check_output
    process = Popen(stdout=PIPE, *popenargs, **kwargs)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1334, in _execute_child
    child_exception = pickle.loads(data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 1382, in loads
    return Unpickler(file).load()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 858, in load
    dispatch[key](self)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 966, in load_string
    raise ValueError, "insecure string pickle"
ValueError: insecure string pickle
>>> 
```

ivankravets, 2016-01-03:
Please try http://docs.platformio.org/en/latest/installation.html#d-development-version

commit fdff35e3 (2016-01-03): Fix package "shasum" validation on Mac OS X 10.11.2 // Resolve #429
closed 2016-01-03 by ivankravets