platformio/platformio-core #430: "platformio lib" doesn't show source url

karlp, 2016-01-03:
```platformio lib search json``` shows you only the vague descriptions from library.json files, with library "64" for instance being, ""Benoit Blanchon": An elegant and efficient JSON library for embedded systems"

```platformio lib show 64``` won't show you anything at all until you've installed it.  This is... not very nice.  Where is it going to be installed from?  What code is this?

Even after installing it, ``... show 64``` still only shows the _author's_ provided url, not the source url at all.  In this case for instance, you get:

```
(.env27)karlp@teros:~/src/hoho$ platformio lib show 64
Json
----
Authors: Benoit Blanchon http://blog.benoitblanchon.fr
Keywords: json, rest, http, web
Frameworks: arduino
Platforms: atmelavr, atmelsam, timsp430, titiva, teensy, freescalekinetis, ststm32, nordicnrf51, nxplpc, espressif, siliconlabsefm32
Version: ab2502f7b8

An elegant and efficient JSON library for embedded systems

(.env27)karlp@teros:~/src/hoho$
```

Benoit's blog is interesting, but it's not useful to say where the code is coming from.  The git version is included, but it _should_ be including the source url, or some form of the entire "repository" section of the library.json block.

ivankravets, 2016-01-03:
How about http://platformio.org/#!/lib/show/64/Json ?

Yes, CLI isn't informative. Do you have propositions?

ivankravets, 2016-01-03:
How about `platformio lib info %ID%` ? I hope that in PlatformIO 3.0 we will allow to use names instead IDs.

karlp, 2016-01-04:
I'd like show or info to be a single command, showing info including the installed state if possible, I don't see any real benefit in having a separate command to see the full details of a package that has been installed vs one that hasn't. 

Thanks for the prompt response!

ivankravets, 2016-12-28:
See example:
* http://docs.platformio.org/en/latest/userguide/lib/cmd_show.html

commit e7cc94c2 (2016-12-28): Show detailed info about a library using `pio lib show` command // Resolve #430
closed 2016-12-28 by ivankravets