platformio/platformio-core #432: Library dependency detection should detect library required by cpp files

sarfata, 2016-01-04:
The [current approach](https://github.com/platformio/platformio/blob/develop/platformio/builder/tools/platformio.py#L295) to detect which libraries are required by another library only walks through `.h` files and not `.cpp` files.

This means that a some dependencies can be missed.

In my case, I have a library (in `<project>/lib`) that uses `ILI9341_t3.h` in the header and this correctly adds this library to the build path. But code in a `.cpp` file that requires `font_Nunito.h` (from `ILI9341_fonts`) does not compile because the cpp file's include statements are not parsed.

A quick and dirty fix for this problem is to add all the includes from other libraries in at least one `.h` file. 

I think this is a bug and should be fixed because it will confuse people and it is pretty hard to diagnose.

As a side note, I would personally be in favor of a declarative dependency systems where each project declares the list of libraries it requires in `platformio.ini` instead of auto-detecting it amongst system-installed dependencies.

ivankravets, 2016-01-04:
PlatformIO scans all source files. See https://github.com/platformio/platformio/blob/develop/platformio/builder/tools/platformio.py#L26:L27

I'll test it, looks like a bug.

> As a side note, I would personally be in favor of a declarative dependency systems where each project declares the list of libraries it requires in platformio.ini instead of auto-detecting it amongst system-installed dependencies.

Do you mean this? https://github.com/platformio/platformio/issues

ivankravets, 2016-01-04:
I've got you. This isn't a bug. Looks like you need to enable http://docs.platformio.org/en/latest/projectconf.html#lib-dfcyclic

sarfata, 2016-01-04:
You are always so fast to reply!

I will try this but I was definitely getting library included from .h files so there is still something suspicious here.

> Do you mean this? https://github.com/platformio/platformio/issues
Which issue did you mean to link to?

ivankravets, 2016-01-04:
> You are always so fast to reply!

Thanks! Support should be excellent without dependency on project type (free or commercial).

> I will try this but I was definitely getting library included from .h files so there is still something suspicious here.

Currently Library Dependency Finder doesn't is turned off by default to avoid a lot of conflicts between installed libs. The safe way is to `#include <nested library>` within `src` files in the project.

> Which issue did you mean to link to?

Sorry. See https://github.com/platformio/platformio/issues/413

-------------

In any case, I converted this issue to `feature`. You will have 3 options: 

1. Directly include dependent library in `src` files
2. Use `lib_require` https://github.com/platformio/platformio/issues/413
3. Create [library.json](http://docs.platformio.org/en/latest/librarymanager/config.html) in `<project>/lib/mylib` and specify [dependencies](http://docs.platformio.org/en/latest/librarymanager/config.html#dependencies)

https://github.com/platformio/platformio/issues/413

sarfata, 2016-02-11:
Just wanted to get back on this and confirm the bug. I have [created a simple project to reproduce ](https://github.com/sarfata/demobug-platformio-432) this issue.

If you try building this project, the build will fail:
````
arm-none-eabi-g++ -o .pioenvs/teensy31/FakeDisplay/FakeDisplay.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -MMD -DF_CPU=72000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX256__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/teensy31/FrameworkArduino -I.pioenvs/teensy31/FakeDisplay -I.pioenvs/teensy31/SPI .pioenvs/teensy31/FakeDisplay/FakeDisplay.cpp
arm-none-eabi-g++ -o .pioenvs/teensy31/SPI/SPI.o -c -fno-rtti -fno-exceptions -std=gnu++0x -felide-constructors -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -MMD -DF_CPU=72000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH -D__MK20DX256__ -DARDUINO=10600 -DTEENSYDUINO=127 -I.pioenvs/teensy31/FrameworkArduino -I.pioenvs/teensy31/FakeDisplay -I.pioenvs/teensy31/SPI .pioenvs/teensy31/SPI/SPI.cpp
.pioenvs/teensy31/FakeDisplay/FakeDisplay.cpp:2:25: fatal error: font_Nunito.h: No such file or directory
#include <font_Nunito.h>
^
compilation terminated.
scons: *** [.pioenvs/teensy31/FakeDisplay/FakeDisplay.o] Error 1
====================================================== [ ERROR ] Took 0.81 seconds ======================================================
````

Remarks:
 - If you uncomment the [`#include <font_Nunito.h>`](https://github.com/sarfata/demobug-platformio-432/blob/master/lib/FakeDisplay/FakeDisplay.h#L3) in `lib/FakeDisplay/FakeDisplay.h` the project builds. This confirms that platformio scans the headers of libraries but not [the cpp files](https://github.com/sarfata/demobug-platformio-432/blob/master/lib/FakeDisplay/FakeDisplay.cpp#L2).
 - If you [enable cyclic dependencies](https://github.com/sarfata/demobug-platformio-432/blob/master/platformio.ini#L27) (in `platformio.ini`), the project builds

What I expected:
 - According to the doc, when cycling dependencies are off, no source files of the libraries should be scanned. The build should fail even if the library is listed in the headers.

ivankravets, 2016-06-22:
@sarfata please move [this line](https://github.com/sarfata/demobug-platformio-432/blob/master/lib/FakeDisplay/FakeDisplay.cpp#L2) to [this file](https://github.com/sarfata/demobug-platformio-432/blob/master/lib/FakeDisplay/FakeDisplay.h#L1) and PlatformIO will automatically find dependency without `lib_dfcyclic`

commit f8db1d11 (2016-06-22): New Library Build System: intelligent dependency finder that interprets C Preprocessor conditional macros // Resolve #432
closed 2016-06-22 by ivankravets