platformio/platformio-core #434: Update framework-arduinoteensy

gefla, 2016-01-04:
Hi,

the framework-arduinoteensy package is out of date. It is older than at least https://github.com/PaulStoffregen/cores/commit/f2948bd19530a2298edcba315e888d35325cc0d0, which is required for DMA UART operation.

As a side note, I have tried finding documentation on the package system and determine
* what source the package was built from
* how to send an update myself

Did I just miss the docs?

     Goodbyte, Gerd.

ivankravets, 2016-01-04:
@gefla thanks! I'll update `framework-arduinoteensy`. If you like PlatformIO, please give us a :star: https://github.com/platformio/platformio/stargazers

ivankravets, 2016-01-04:
@gefla please `platformio update`.

closed 2016-01-04 by ivankravets

ivankravets, 2016-01-04:
@PaulStoffregen could you make release/tag for https://github.com/PaulStoffregen/cores in future? I've just subscribed to your repository and will be notified for the new releases. Thanks a lot!

gefla, 2016-01-04:
As mentioned on gitter: Done, works.

PaulStoffregen, 2016-01-04:
I generally don't use github to release Teensyduino.  I'll try to tag releases, but odds are strong I'll tend to neglect that step.  My release process already involves a lot of non-github steps!

ivankravets, 2016-01-05:
@PaulStoffregen Thanks! I'll try to monitor it manually too.

commit 7a7fa845 (2016-01-04): Update Teensy Arduino Framework to 1.26 // Resolve #434