platformio/platformio-core #437: library.json 'exclude' option is not used when building libraries in 'lib/'

sarfata, 2016-01-05:
I have a library in 'lib/' and have added some unit tests to it but there does not seem to be a way to exclude them from the build.

The `exclude` option in a `library.json` is not used (I don't even see if that file is parsed for local libraries in the source) and the `src_filter` option is not applied to libraries.

Is there another solution I missed?

ivankravets, 2016-01-05:
> The exclude option in a library.json is not used (I don't even see if that file is parsed for local libraries in the source) and the src_filter option is not applied to libraries.

See https://github.com/platformio/platformio/issues/289

> Is there another solution I missed?

Sure, just place your source code to `src` folder in the library and keep `test` folder on the same level as `src` folder. See example with "Bar" library https://github.com/platformio/platformio/blob/develop/platformio/commands/init.py#L197:L202

ivankravets, 2016-01-11:
Please reopen issue if you have any questions.

closed 2016-01-11 by ivankravets

sarfata, 2016-02-11:
This worked perfectly. Thank you.