platformio/platformio-core #438: Chipkit PIC32 boards support

deladriere, 2016-01-05:
I would love to have PIC32 boards supported, thanks
How come they are all absent from Platformio?

deladriere, 2016-01-06:
Sorry but I can't find how to apply labels to this issue (board ...)
could someone do it ?
thx

deladriere, 2016-02-20:
PIC32 are now supported by Arduino IDE. Maybe now it could also workk on PIO ?
see : https://raw.githubusercontent.com/chipKIT32/chipKIT-core/master/package_chipkit_index.json

adilinden, 2016-03-05:
Yes, would be nice to see chipKIT32 support.  It would make my day :)

ivankravets, 2016-04-27:
@deladriere , @adilinden  please re-test the latest http://docs.platformio.org/en/latest/installation.html#development-version

ivankravets, 2016-07-25:
Final firmware doesn't work. I reopen issue.

commit d5b41fe6 (2016-04-27): Add support for Microchip PIC32 development platform // Resolve #438
commit c61ff611 (2016-07-28): Improve support for Microchip PIC32 development platform and ChipKIT boards // Resolve #438
closed 2016-07-28 by ivankravets