platformio/platformio-core #439: Support UDOO boards

gefla, 2016-01-06:
Two board classes:
* [Neo](http://www.udoo.org/udoo-neo/) (basic/extended/full variants all using Freescale™ i.MX 6SoloX applications processor with an embedded ARM Cortex-A9 core and a Cortex-M4 Core)
* [Dual/Quad](http://www.udoo.org/udoo-dual-and-quad/), using a separate Atmel® SAM3X8E

The manufacturer provides a patched Arduino IDE based on the board support package linked below. There are two firmware upload paths for the Neo:
1. Run [udooneo-m4uploader](https://github.com/ektor5/udooneo-m4uploader) tool on the local Linux core
2. Run [udooclient](https://github.com/UDOOboard/udoofota) on developer workstation, which uses a TCP connection to a service that runs m4uploader locally

For the Dual/Quad variants, the bossac tool is used (patched, see [repository](https://github.com/UDOOboard/bossac)).

My main interest is in the Neo boards. I'm happy to have a go at writing the support, though I would probably need some help with that.

References
 * https://github.com/UDOOboard/arduino-board-package

onliner10, 2016-10-31:
I can see this issue is just hanging there untouched for a long time, so as I'm also very much interested, maybe somebody can give any tips how to do this? If someone can give me some tips, where to start, I'd be glad to pick up this issue.

Thanks in advance

ivankravets, 2016-10-31:
@onliner10 here is docs http://docs.platformio.org/en/latest/platforms/custom_platform_and_board.html

I see that these boards are based on Atmel SAM. In this case, just need to add new board to https://github.com/platformio/platform-atmelsam and integrate uploading.

onliner10, 2016-11-01:
I tried to just add new JSON manifest, but I think it's not as simple since it's not really Atmel SAM board but Freescale i.MX 6SoloX with integrated both M9 and M4 chips. I've went through their documentation on http://www.udoo.org/docs-neo/Advanced_Topics/Recompile_MQX_Libraries.html and it seems like we need some MQX libraries (whatever they are..). When I last tried uploading sketch compiled for due in Platformio, i was unable to upload it (whereas the ones compiled with their patched Arduino IDE uploaded without any issues). So, i tried to just copy all their board files from AppData and create new framework out of it, but I don't really think this is the good way